project-chip/connectedhomeip #4921: Implement the Descriptor Cluster

andy31415, 2021-02-18:
#### Problem
Initial operation and connectivity to a device depends heavly on the other end being able to identify the device type.

We need to implement the Descriptor cluster (section 7.4.6 currently in the CHIP Spec)

 #### Proposed Solution
 
 Implement the new cluster. Likely need incremental implementations since the cluster relies on struct and list data types which are not yet supported by our code generators.

andy31415, 2021-02-18:
Document on how to add clusters to CHIP: https://docs.google.com/document/d/1_FoSrDz89QYhIn24m-6oRyabspJujkBP695p_MMhI5M/edit#

yufengwangca, 2021-02-25:
This is implemented in https://github.com/project-chip/connectedhomeip/pull/5020

closed 2022-01-25 by woody-apple