project-chip/connectedhomeip #4933: Chip-tool: To log messages for all phases of the provisioning flow

cjandhyala, 2021-02-19:
Need chip-tool to log all the phases of the provisioning flow as shown in Fig 26 of section 5.3. Since most of the messages exchanged during these phases are encrypted messages, we need to have verbose logs to make sure device completed all the phases mentioned in the provisioning flow , for TE1

woody-apple, 2021-02-19:
Re-titling. Definitely needed for TE1, but not a CHIP tool issue. This is a request to log in libCHIP.

andy31415, 2021-03-10:
Could you provide a list of logs that you would expect (some sort of mock logs or similar)?

Could you also attach an image  (or asciiart?) what "Fig 26 of section 5.3" means? It would help whoever picks this up understand the issue faster without needing to follow extra links/documents.

cjandhyala, 2021-03-12:
We need log messages that each of the phases in the commissioning flow is entered/completed.

[image]

Concurrent connection commissioning flow
[image]

Non-concurrent commissioning flow
[image]

pan-apple, 2021-04-15:
Some updates on this issue:

Currently we are supporting the following stages of commissioning flow. And, the logs have been added to these steps.

1. Discover device on BLE
2. Assign local session key
3. Spake2p handshake.
4. Log session key assigned by the peer
5. Network provisioning (It's not done using the cluster API yet)

#### Note: The logs provided below are interspersed with other logs.
#### Note: The logs from the network provisioning might change as we bring in network provisioning cluster.

For example, these logs are from the commissioner application:
```
CHIP: [BLE] Connecting to device with discriminator: 3840
CHIP: [BLE] Assigned local session key ID 95
CHIP: [BLE] Sent PBKDF param request
CHIP: [BLE] Received PBKDF param response
CHIP: [BLE] Sent spake2p msg1
CHIP: [BLE] Received spake2p msg2
CHIP: [BLE] Peer assigned session key ID 0
CHIP: [BLE] Sent spake2p msg3
CHIP: [IN] New pairing for device 0x00000000000000a0, key 0!!
CHIP: [CTL] Remote device completed SPAKE2+ handshake
CHIP: [NP] Sending Network Creds
CHIP: [NP] Received kIPAddressAssigned
CHIP: [CTL] Remote device was assigned an ip address
```

The following are the corresponding logs from the device
```
chip[BLE]: Assigned local session key ID 0
chip[BLE]: Waiting for PBKDF param request
chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable, device name CHIP-3840)
chip[DL]: Device already advertising, stop active advertisement and restart
all-clusters-app: Short Manual(decimal) setup code: 35767807533
all-clusters-app: Long Manual(decimal) setup code:  757678075309050652793
all-clusters-app: QR CODE Text: 'CH:H34DV*-00 0C9SS0'
chip[DL]: BLE GAP connection established (con 0)
chip[BLE]: Received PBKDF param request
chip[BLE]: Sent PBKDF param response
chip[BLE]: Received spake2p msg1
chip[BLE]: Peer assigned session key ID 97
chip[BLE]: Sent spake2p msg2
chip[BLE]: Received spake2p msg3
chip[IN]: New pairing for device 0x0000000000000000, key 97!!
chip[SVR]: Device completed SPAKE2+ handshake
chip[NP]: Received kWiFiAssociationRequest.
chip[SVR]: Device was assigned network credentials
chip[DIS]: CHIP minimal mDNS started advertising.
chip[DIS]: Replying to DNS-SD service listing request
chip[NP]: Sending IP Address
chip[DIS]: CHIP minimal mDNS started advertising.
```

pan-apple, 2021-04-15:
@cjandhyala, I updated the issue with the logs that are currently available as of 672cf76841517c07b180fd8d157733bd71eab3d7 commit.

Could you help identify additional logs that are needed for TE2? The complete commissioning flow will not be supported for TE2. So we can add the ones that should be supported for current milestone, and open a new issue for future milestones.

cjandhyala, 2021-04-15:
Thx @pan-apple , Sure, Could you let know on what parts of the commissioning flow supported so far i.e. for TE2, so that we can make identify for the required log statement.

pan-apple, 2021-04-15:
Looking at the TE2 deliverables, the following protocols should also add the relevant log messages.
1. DNS-SD
2. mDNS discovery
3. CASE state machine
4. Operational Credential provisioning

PR #6093 (2021-04-16): [app] add logging for dns-sd server

 #### Problem

Logs are missing for the dns-sd server.

 Partially fixes #4933
closed 2021-04-16 by woody-apple