project-chip/connectedhomeip #4935: nrfconnect lock-app chip device fails to boot with Unaligned memory access error

RoyAnupam, 2021-02-19:
**Problem:** nrfconnect lock-app device fails to boot with Unaligned memory access error

 * expected behavior: nRF52840-DK chip device should boot normally
 
* actual behavior: nRF52840-DK chip device fails to boot normally and keeps continuously re-booting with Unaligned memory access error

 * steps to reproduce: 
       1. build connectedhomeip/examples/lock-app/nrfconnect for nRF52840-DK board
       2. Flash zephyr.hex file in nRF52840-DK with nRF Connect v3.5.0 Windows application
       3. Power up
 
 * system configuration: lock-app/nrfconnect for nRF52840-DK board

 * screenshots, images, logs, etc.
 
nRF5280-DK board fails to boot normally and keeps reboot automatically with Error messages

**Boot Log**
Z+
W+▒ещ5
I: 8 Sectors of 4096 bytes
I: alloc wra: 0, ff0
I: data wra: 0, 0
*** Booting Zephyr OS version 2.4.99  ***
I: Init CHIP stack
I: SoftDevice Controller build revision: 
I: e5 c7 9c d9 91 00 1d 66 |.......f
I: ea fb 6e 7b 98 2f 42 0d |..n{./B.
I: f1 60 93 c8             |.`..    
I: Starting CHIP task
I: Init Thread stack
I: 496[DL] OpenThread started: OK
I: 499[DL] Setting OpenThread device type to MINIMAL END DEVICE
I: 505[ZCL] Using ZAP configuration...
I: 508[ZCL] deactivate report event
D: 511[IN] TransportMgr initialized
I: 515[IN] local node id is lu

D: 518[DL] SetAdvertisingEnabled(true)
**E: ***** USAGE FAULT *****
E:   Unaligned memory access**
E: r0/a1:  0x2000583d  r1/a2:  0x0006f8e6  r2/a3:  0x0000001a
E: r3/a4:  0x0006f901 r12/ip:  0xffffffff r14/lr:  0x00048e8f
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x0004ad56
**E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0**
E: Current thread: 0x20002eb0 (main)
E: Resetting systemI: nRF5 802154 radio initialized
I: 8 Sectors of 4096 bytes
I: alloc wra: 0, ff0
I: data wra: 0, 0

Kindly suggest what could be the issue and how to resolve it? Thank You very much

Damian-Nordic, 2021-02-23:
@RoyAnupam which revision of CHIP did you use for testing? The issue looks very similar to what we encountered recently and fixed in https://github.com/project-chip/connectedhomeip/commit/53a2046cb20588c84ce91b754a0d32315438afcd. Could you check if the problem remains in the current version?

RoyAnupam, 2021-03-02:
Hello @Damian-Nordic , Completely missed it. Apology. I will check & update

RoyAnupam, 2021-03-02:
Hello @Damian-Nordic ,
Thank you for your guide. After pulling latest, it seems, unaligned memory access is resolved, however I ran into different problem while provisioning lock-app accessory to THREAD network.

Following is the error in UART terminal for the nRF chip device. The error appeared while in *pairing stage* with Android Mobile (Provisioner). Could you suggest if I am missing something.
==========================================================================
I: 18989 [BLE]Received pairing message for lu
E: 18993 [SVR]OnRendezvousStatusUpdate: Error 4042 (0x00000FCA)
I: 18999 [SVR]OnRendezvousError: Error 4042 (0x00000FCA)
E: 19004 [SVR]OnRendezvousStatusUpdate: Error 4042 (0x00000FCA)
D: 19010 [DL]SetAdvertisingEnabled(false)
I: 19014 [SVR]Persisting admin ID 0, next available 1
I: 19025 [SVR]Persisting admin ID successfully
I: 19029 [BLE]Releasing end point's BLE connection back to application.
I: 19035 [SVR]OnRendezvousError: Error 4042 (0x00000FCA)
I: 19040 [SVR]Persisting admin ID 0, next available 1
I: 19052 [SVR]Persisting admin ID successfully
D: 19056 [DL]In DriveBLEState
D: 19126 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
E: 19134 [BLE]no endpoint for BLE sent data ack
D: 21595 [DL]Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
I: 21602 [BLE]Releasing end point's BLE connection back to application.
E: 21609 [BLE]HandleChipConnectionReceived failed, err = 6018
E: 21614 [BLE]failed handle new chip BLE connection, status = 6018
==========================================================

LuDuda, 2021-03-04:
@RoyAnupam I believe this is due to some incompatibility between CHIP versions on Android and accessory side. After I rebuilt Android CHIP Tool from the master, the problem disappeared.

RoyAnupam, 2021-03-14:
@LuDuda , Okay. I will try rebuilding latest CHIP tool from master & give it a try. Thanks for your suggestion.

LuDuda, 2021-03-17:
At least I take advantage of the docker image which is used in our CI to have all required toolchain.

1. Run the `chip-build-android` docker. Note to replace `~/git/connectedhomeip` with your location of the CHIP SDK.

```
docker run -it -e RUNAS=$(id -u) -v ~/git/connectedhomeip:/var/chip connectedhomeip/chip-build-android
```

2. Inside docker execute the following commands:
```
cd /var/chip
yes | "$ANDROID_HOME"/tools/bin/sdkmanager --licenses
TARGET_CPU=arm64 ./scripts/examples/android_app.sh
cd src/android/CHIPTool
./gradlew build
```

3. The debug Android package `app-debug.apk` will be generated at `./app/build/outputs/apk/debug/`.

4. Install the application on the Android phone.
a) Install the Android Debug Bridge (adb) package by running the following command:
`sudo apt install android-tools-adb`
b) Enable USB debugging on the smartphone. See the Configure on-device developer options guide on the Android Studio hub for detailed information. If the Install via USB option is supported for your Android version, turn it on.
c) Plug the smartphone into an USB port on your PC.
d) Run the following command to install the application:
`adb install -r -t app-debug.apk`
e) Navigate to Android Settings and grant Camera and Loecation permissions to CHIP Tool.

RoyAnupam, 2021-03-18:
hello @LuDuda ,

>  Note to replace ~/git/connectedhomeip with your location of the CHIP SDK.

 Sorry, i got little disconnected here. By CHIP SDK, do you mean, connectedhomeip project root directory? Could not understand fully .

LuDuda, 2021-03-18:
This is the only repository we have with CHIP Tool for Android :) So yes! 😄

RoyAnupam, 2021-03-18:
@LuDuda ,
 Thank you so much for your guidance. Indeed, now with help of docker, I am able to pass through previous javac error while running android_app.sh script. However, while doing gradlew build, I met following error.

$./gradlew build
===============================================
  public void log(LogLevel level, String region, String msg) {
                  ^
  symbol:   class LogLevel
  location: class NativeCommissionerLogger
28 errors

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 13m 56s
13 actionable tasks: 1 executed, 12 up-to-date
=============================================

LuDuda, 2021-03-18:
I guess you missed to pull submodules:
`cd /var/chip`
`git submodule update --init`

I tried yesterday, and it worked, also CI indicates it.

RoyAnupam, 2021-03-18:
@LuDuda ,
 Unfortunately, I am still facing the same issue even after running submodule update.
In the process, following is one observation I have, which i wish to share.

After running below script, there is 'pw not found' error.
======================================================
$TARGET_CPU=arm64 ./scripts/examples/android_app.sh
.....
.......
.............
Checking the environment:
+++ '[' -z '' ']'
+++ echo ''

+++ '[' -z '' ']'
+++ pw --no-banner --loglevel info doctor
/var/chip1/.environment/activate.sh: line 153: **pw: command not found**

==========================================================

Could above issue be related?
By the way, I am using Ubuntu 20.04.1 LTS, just for your information, if there is anything possibly related with it too.
Thank You

LuDuda, 2021-03-24:
Uhh.. Final try.. let's call this before building commands:
```
source scripts/bootstrap.sh
source scripts/activate.sh
```

woody-apple, 2022-01-25:
This appears resolved

closed 2022-01-25 by woody-apple