project-chip/connectedhomeip #4937: raspi : hung-up on in network provisioning

cjandhyala, 2021-02-19:
Raspi hung up in network provisioning. Per spec, once the device provisioned with network credentials ,device BLE GAP connection should be terminated. 

chip-device-ctrl >  connect -ble 3840 12345678
CHIP: [BLE] NewConnection
CHIP: [BLE] CBManagerState: ON
CHIP: [BLE] Connecting to device with discriminator: 3840
CHIP: [BLE] Warning: OnBleConnectionComplete is called multiple times for the same peripheral.
CHIP: [BLE] subscribe complete, ep = 0x111e7a878
CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
CHIP: [BLE] using BTP fragment sizes rx 20 / tx 128.
CHIP: [BLE] local and remote recv window size = 3
CHIP: [BLE] Sent PBKDF param request
CHIP: [BLE] Received pairing message from 12344321
CHIP: [BLE] Received pairing message for 112233
CHIP: [BLE] Received PBKDF param response
CHIP: [BLE] Sent spake2p msg1
CHIP: [BLE] Received pairing message from 12344321
CHIP: [BLE] Received pairing message for 112233
CHIP: [BLE] Received spake2p msg2
CHIP: [BLE] Sent spake2p msg3
CHIP: [IN] New pairing for device 12344321, key 0!!
CHIP: [CTL] Remote device completed SPAKE2+ handshake

CHIP: [NP] Sending Network Creds. Delegate 0x7f9ee101ea18

CHIP: [IN] Secure message was encrypted: Msg ID 0
CHIP: [IN] Sending msg from 112233 to 12344321
CHIP: [IN] Sending secure msg on connection specific transport
CHIP: [IN] Secure msg send status 0

--------------Device Side log--------------

CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Received rendezvous message for 12344321
CHIP:BLE: Received rendezvous message from 112233
CHIP:NP: Received kWiFiAssociationRequest.

CHIP:NP: LinuxNetworkProvisioningDelegate: SSID: Bewareofpython
CHIP:DL: wpa_supplicant: added network: SSID: Bewareofpython: /fi/w1/wpa_supplicant1/Interfaces/0/Networks/0
CHIP:DL: wpa_supplicant: connected to network: SSID: Bewareofpython
CHIP:DL: wpa_supplicant: failed to save config: GDBus.Error:fi.w1.wpa_supplicant1.UnknownError: Not allowed to update configuration (update_config=0)
CHIP:SVR: Device was assigned network credentials
CHIP:SVR: Device completed Rendezvous process
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70
CHIP:BLE: Indication confirmation, 0xffffa4023f70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffffa4023f70

yunhanw-google, 2021-02-24:
central is running in macbook

erjiaqing, 2021-02-24:
Seems Linux does not send IP address to the device controller which is required by current rendezvous session.

cjandhyala, 2021-03-01:
Damian commented that It seems the ConnectivityManagerImpl of Linux needs to implement UpdateInternetConnectivityState handling the kWiFiStationState_Connected event , hence the controller is not aware of the IP address and continuing the BT connection. Yunhan/Song, Could you look into this ?

closed 2021-03-03 by andy31415