project-chip/connectedhomeip #4947: chip-tool (chip-tool-server) linking errors on Android version < = L

mdamle, 2021-02-19:
Running chip-tool (chip-tool-server) on devices with Android version < = L causes linking (dynamic loader) issues.

```
WARNING: linker: chip-tool: unused DT entry: type 0x6ffffef5 arg 0x30c8 
WARNING: linker: chip-tool: unused DT entry: type 0x6ffffffe arg 0x3574 
WARNING: linker: chip-tool: unused DT entry: type 0x6fffffff arg 0x1 
CANNOT LINK EXECUTABLE: empty/missing DT_HASH in "chip-tool" (built with --hash-style=gnu?) 
```
Compilation command:
```
gn gen out/android_arm --args='target_os="android" target_cpu="arm" chip_build_tests=false chip_device_platform="linux" chip_enable_mdns=false chip_enable_ble=false chip_enable_openthread=false chip_enable_wifi=false chip_bypass_rendezvous=true android_ndk_root="/Users/makdam/Library/Android/android-ndk-r21d" android_sdk_root="/Users/makdam/Library/Android/sdk"'
```

Reason seems to be: https://android.googlesource.com/platform/bionic/+/master/android-changes-for-ndk-developers.md#gnu-hashes-availible-in-api-level-23 
With API level > =23, dynamic linker switched to using --hash-style=gnu whereas older Android versions (<= L) use --hash-style=sysv.

**Possible solution**:
Append --hash-style=both to ldflags of the complier. As an example:
```
diff --git a/build/config/compiler/BUILD.gn b/build/config/compiler/BUILD.gn
index d018af51..7247f89f 100644
--- a/build/config/compiler/BUILD.gn
+++ b/build/config/compiler/BUILD.gn
@@ -177,6 +177,8 @@ config("strict_warnings") {
   if (current_os == "linux" || current_os == "android") {
     ldflags += [ "-Wl,-z,defs" ]
+    # For Android OS <= Android L, we need to add
+    ldflags += [ "--hash-style=both" ]
   }
 }
```

Doing this however leads to a compilation failure:
```
clang++: error: unsupported option '--hash-style=both'
[4/15] ld ./chip-shell
FAILED: chip-shell 
```

mdamle, 2021-02-19:
Using this tweak to build/config/compiler/BUILD.gn
```
$ git diff build/config/compiler/BUILD.gn
diff --git a/build/config/compiler/BUILD.gn b/build/config/compiler/BUILD.gn
index d018af51..09886e05 100644
--- a/build/config/compiler/BUILD.gn
+++ b/build/config/compiler/BUILD.gn
@@ -93,6 +93,10 @@ config("abi_default") {
     }
   }
   ldflags = cflags
+  if (current_os == "android") {
+    ldflags += [ "-Wl,--hash-style=both" ]
+  }
+  print(current_os, ldflags)
 }
```
leads to a different linking error.
```
# chip-tool                                                       
WARNING: linker: chip-tool: unused DT entry: type 0x6ffffef5 arg 0x3094
WARNING: linker: chip-tool: unused DT entry: type 0x6ffffffe arg 0x3c28
WARNING: linker: chip-tool: unused DT entry: type 0x6fffffff arg 0x1
WARNING: linker: libc++_shared.so: unused DT entry: type 0x6ffffef5 arg 0x25828
WARNING: linker: libc++_shared.so: unused DT entry: type 0x6ffffffe arg 0x2fc28
WARNING: linker: libc++_shared.so: unused DT entry: type 0x6fffffff arg 0x2
CANNOT LINK EXECUTABLE: cannot locate symbol "__register_atfork" referenced by "chip-tool"...
```

mspang, 2021-02-21:
Can you submit the fix in the form of a pull request?

mspang, 2021-02-21:
Oh, per the last line it doesn't work :(

mspang, 2021-02-22:
https://github.com/android/ndk/issues/964

```
## `__register_atfork` (Available in API level >= 23)

To allow `atfork` and `pthread_atfork` handlers to be unregistered on
`dlclose`, the implementation changed in API level 23. Unfortunately this
requires a new libc function `__register_atfork`. Code using these functions
that is built with a target API level >= 23 therefore will not load on earlier
versions of Android, with an error referencing `__register_atfork`.

*Resolution*: build your code with an NDK target API level that matches your
app's minimum API level, or avoid using `atfork`/`pthread_atfork`.
```

mdamle, 2021-02-22:
Ok so following set of three changes seem to work. The first one seems non controversial, for the second one, we need to find an elegant way to specify a build time overridable way for api-level and for the third one we need to find a proper fix.

1. Append --hash-style=both to ldflags of the complier.
```
$ git diff build/config/compiler/BUILD.gn
diff --git a/build/config/compiler/BUILD.gn b/build/config/compiler/BUILD.gn
index d018af51..09886e05 100644
--- a/build/config/compiler/BUILD.gn
+++ b/build/config/compiler/BUILD.gn
@@ -93,6 +93,10 @@ config("abi_default") {
     }
   }
   ldflags = cflags
+  if (current_os == "android") {
+    ldflags += [ "-Wl,--hash-style=both" ]
+  }
+  print(current_os, ldflags)
 }
```
2. Change android api-level to 21
```
$ git diff build/toolchain/android/BUILD.gn
diff --git a/build/toolchain/android/BUILD.gn b/build/toolchain/android/BUILD.gn
index c9e6039d..cabcadf3 100644
--- a/build/toolchain/android/BUILD.gn
+++ b/build/toolchain/android/BUILD.gn
@@ -17,7 +17,7 @@ import("android_toolchain.gni")
 
 android_clang_toolchain("android_arm") {
   # TODO: Drop API level to 21 after getifaddrs issue fixed
-  api_level = "24"
+  api_level = "21"
   toolchain_args = {
     current_cpu = "arm"
   }
```
3. Since dropping api level to 21 exposes an existing issue with getifaddrs, comment out the relevant code
```
$ git diff src/platform/Linux/ConfigurationManagerImpl.cpp
diff --git a/src/platform/Linux/ConfigurationManagerImpl.cpp b/src/platform/Linux/ConfigurationManagerImpl.cpp
index 57090c6d..f133dd4c 100644
--- a/src/platform/Linux/ConfigurationManagerImpl.cpp
+++ b/src/platform/Linux/ConfigurationManagerImpl.cpp
@@ -76,28 +76,28 @@ exit:
 
 CHIP_ERROR ConfigurationManagerImpl::_GetPrimaryWiFiMACAddress(uint8_t * buf)
 {
-    struct ifaddrs * addresses = NULL;
-    CHIP_ERROR error           = CHIP_NO_ERROR;
-    bool found                 = false;
+    // struct ifaddrs * addresses = NULL;
+    CHIP_ERROR error = CHIP_NO_ERROR;
+    /*     bool found                 = false;
 
-    VerifyOrExit(getifaddrs(&addresses) == 0, error = CHIP_ERROR_INTERNAL);
-    for (auto addr = addresses; addr != NULL; addr = addr->ifa_next)
-    {
-        if ((addr->ifa_addr) && (addr->ifa_addr->sa_family == AF_PACKET) && strncmp(addr->ifa_name, "lo", IFNAMSIZ) != 0)
+        VerifyOrExit(getifaddrs(&addresses) == 0, error = CHIP_ERROR_INTERNAL);
+        for (auto addr = addresses; addr != NULL; addr = addr->ifa_next)
         {
-            struct sockaddr_ll * mac = (struct sockaddr_ll *) addr->ifa_addr;
-            memcpy(buf, mac->sll_addr, mac->sll_halen);
-            found = true;
-            break;
+            if ((addr->ifa_addr) && (addr->ifa_addr->sa_family == AF_PACKET) && strncmp(addr->ifa_name, "lo", IFNAMSIZ) != 0)
+            {
+                struct sockaddr_ll * mac = (struct sockaddr_ll *) addr->ifa_addr;
+                memcpy(buf, mac->sll_addr, mac->sll_halen);
+                found = true;
+                break;
+            }
+        }
+        freeifaddrs(addresses);
+        if (!found)
+        {
+            error = CHIP_ERROR_NO_ENDPOINT;
         }
-    }
-    freeifaddrs(addresses);
-    if (!found)
-    {
-        error = CHIP_ERROR_NO_ENDPOINT;
-    }
 
-exit:
+    exit: */
     return error;
 }
```

Note that I still see a warning while executing chip-tool but it runs without any issues.
```
 # chip-tool                                                       
WARNING: linker: chip-tool: unused DT entry: type 0x6ffffef5 arg 0x3040
WARNING: linker: chip-tool: unused DT entry: type 0x6ffffffe arg 0x3bc8
WARNING: linker: chip-tool: unused DT entry: type 0x6fffffff arg 0x1
WARNING: linker: libc++_shared.so: unused DT entry: type 0x6ffffef5 arg 0x25828
WARNING: linker: libc++_shared.so: unused DT entry: type 0x6ffffffe arg 0x2fc28
WARNING: linker: libc++_shared.so: unused DT entry: type 0x6fffffff arg 0x2
Usage:
  chip-tool cluster_name command_name [param1 param2 ...]

  +-------------------------------------------------------------------------------------+
  | Clusters:                                                                           |
  +-------------------------------------------------------------------------------------+
  | * barriercontrol                                                                    |
  | * basic                                                                             |
  | * binding                                                                           |
  | * colorcontrol                                                                      |
  | * doorlock                                                                          |
  | * groups                                                                            |
  | * identify                                                                          |
  | * levelcontrol                                                                      |
  | * onoff                                                                             |
  | * pairing                                                                           |
  | * payload                                                                           |
  | * reporting                                                                         |
  | * scenes                                                                            |
  | * temperaturemeasurement                                                            |
  +-------------------------------------------------------------------------------------+
```

mspang, 2021-02-24:
Feel free to open a PR for the parts of the fix that already work.

woody-apple, 2022-01-25:
This appears resolved

closed 2022-01-25 by woody-apple