project-chip/connectedhomeip #4949: Python Controller on raspi fails to connect to the device

cjandhyala, 2021-02-19:
Python Controller on raspi (Ubuntu Server) fails to connect to the device (running on another raspi)

chip-device-ctrl > connect -ble 3840 12345678 1
Device is assigned with nodeid = 1
CHIP:DL: TRACE: Bus acquired for name C-0aae
CHIP:DL: TRACE: Bluez mainloop starting BluezMainLoop
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: Long dispatch time: 1000 ms
CHIP:DL: TRACE: Device DC:A6:32:8D:70:74 Service data: {'0000feaf-0000-1000-8000-00805f9b34fb': <[byte 0x00, 0x00, 0x0f, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: TRACE: Found CHIP BLE Device: 3840
CHIP:DL: Device already discovered
CHIP:DL: FAIL: NULL service in BluezConnectionInit
CHIP:DL: New BLE connection 0xffff8c004df0, device DC:A6:32:8D:70:74, path /org/bluez/hci0/dev_DC_A6_32_8D_70_74
CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.NotAvailable: Operation currently not available
CHIP:DL: HandlePlatformSpecificBLEEvent 16385
CHIP:DL: C1 is NULL in SendWriteRequestImpl
CHIP:DL: Bluez disconnected
CHIP:DL: Bluez notify CHIPoBluez connection disconnected
Failed to establish secure session to device: 6012
Device temporary node id (**this does not match spec**): 1
chip-device-ctrl > 

$ hciconfig
hci0:	Type: Primary  Bus: UART
	BD Address: DC:A6:32:7E:5C:77  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN ISCAN 
	RX bytes:16052 acl:50 sco:0 events:416 errors:0
	TX bytes:3635 acl:50 sco:0 commands:145 errors:0

erjiaqing, 2021-02-24:
According to the log output,
```
CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.NotAvailable: Operation currently not available
```

The error is on the Bluez / Linux bluetooth implementation side, and is likely not related to the Python binding, change the title to "BLE central on raspi fails to connect to the device"

erjiaqing, 2021-02-24:
On my side, it says

```
hci0:   Type: Primary  Bus: UART
        BD Address: DC:A6:32:9E:2E:A7  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING 
        RX bytes:4086447 acl:2787 sco:0 events:128083 errors:0
        TX bytes:55693 acl:2787 sco:0 commands:1111 errors:0
```

Can you try to run on the rpi runing controller?

```
sudo hciconfig -i hci0 power off
sudo hciconfig -i hci0 bredr off
sudo hciconfig -i hci0 power on
```

yunhanw-google, 2021-02-24:
@Damian-Nordic, could you take a look, we use device controller in mac, which can make ble connection successfully with chip-tool-server in raspi, but when run ble central c implementation in raspi, it is occasionally failing

Damian-Nordic, 2021-03-04:
@yunhanw-google yes, I'm setting up everything on RPi as on my laptop it works pretty smoothly.

cjandhyala, 2021-03-25:
not seeing this issue any more

closed 2021-03-25 by cjandhyala