project-chip/connectedhomeip #4965: crash in OnBleEndPointConnectionComplete

mspang, 2021-02-23:
```
Restarting bluetooth (via systemctl): bluetooth.service.
yuif:32573:~/connectedhomeip2% gdb --args /home/spang/connectedhomeip2/.environment/pigweed-venv/bin/python ./.environment/pigweed-venv/bin/chip-device-ctrl
GNU gdb (Debian 10.1-1.7) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/spang/connectedhomeip2/.environment/pigweed-venv/bin/python...
(No debugging symbols found in /home/spang/connectedhomeip2/.environment/pigweed-venv/bin/python)
(gdb) r
Starting program: /home/spang/connectedhomeip2/.environment/pigweed-venv/bin/python ./.environment/pigweed-venv/bin/chip-device-ctrl
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 1129235]
[New Thread 0x7ffff5ca0700 (LWP 1129237)]
[New Thread 0x7ffff549f700 (LWP 1129238)]
[New Thread 0x7ffff4c9e700 (LWP 1129239)]
[New Thread 0x7fffe7fff700 (LWP 1129240)]
[New Thread 0x7fffe77fe700 (LWP 1129241)]
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233

[New Thread 0x7fffe6ffd700 (LWP 1129242)]
CHIP:DL: CHIP task running
Chip Device Controller Shell

chip-device-ctrl > connect -ble 3840 12345678 9987
Device is assigned with nodeid = 9987
[New Thread 0x7fffe67fc700 (LWP 1129243)]
CHIP:DL: TRACE: Bus acquired for name C-3b0f
CHIP:DL: TRACE: Bluez mainloop starting BluezMainLoop
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: Long dispatch time: 1000 ms
CHIP:DL: StartDiscovery complete
CHIP:DL: TRACE: Device 44:50:EB:5B:7E:D2 Service data: {'0000fe50-0000-1000-8000-00805f9b34fb': <[byte 0x83, 0x59]>}
CHIP:DL: TRACE: Device DC:A6:32:8C:93:C3 Service data: {'0000feaf-0000-1000-8000-00805f9b34fb': <[byte 0x00, 0x00, 0x0f, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: TRACE: Found CHIP BLE Device: 3840
CHIP:DL: FAIL: NULL service in BluezConnectionInit
CHIP:DL: New BLE connection 0x7fffec007830, device DC:A6:32:8C:93:C3, path /org/bluez/hci0/dev_DC_A6_32_8C_93_C3
CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.NotAvailable: Operation currently not available
CHIP:DL: HandlePlatformSpecificBLEEvent 16385
CHIP:DL: C1 is NULL in SendWriteRequestImpl
CHIP:DL: Bluez disconnected
CHIP:DL: Bluez notify CHIPoBluez connection disconnected
Failed to establish secure session to device: 6012
Device temporary node id (**this does not match spec**): 9987
chip-device-ctrl > 
chip-device-ctrl > 
chip-device-ctrl > 
chip-device-ctrl > 
chip-device-ctrl > CHIP:DL: TRACE: Device DC:A6:32:8C:93:C3 Service data: {'0000feaf-0000-1000-8000-00805f9b34fb': <[byte 0x00, 0x00, 0x0f, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: TRACE: Found CHIP BLE Device: 3840
CHIP:DL: FAIL: NULL service in BluezConnectionInit
CHIP:DL: New BLE connection 0x7fffec007f20, device DC:A6:32:8C:93:C3, path /org/bluez/hci0/dev_DC_A6_32_8C_93_C3
CHIP:DL: HandlePlatformSpecificBLEEvent 16385
CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.NotAvailable: Operation currently not available
CHIP:DL: C1 is NULL in SendWriteRequestImpl
CHIP:DL: Bluez disconnected
CHIP:DL: Bluez notify CHIPoBluez connection disconnected

Thread 7 "python" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe6ffd700 (LWP 1129242)]
0x00007ffff623b770 in chip::Transport::BLE::OnBleEndPointConnectionComplete (endPoint=0x7ffff634cf88 <chip::Ble::BleEndPointPool::Get(unsigned long) const::sEndPointPool>, err=6012) at ../../src/transport/BLE.cpp:215
215	            ble->mDelegate->OnRendezvousError(err);
(gdb) bt
#0  0x00007ffff623b770 in chip::Transport::BLE::OnBleEndPointConnectionComplete(chip::Ble::BLEEndPoint*, int) (endPoint=0x7ffff634cf88 <chip::Ble::BleEndPointPool::Get(unsigned long) const::sEndPointPool>, err=6012) at ../../src/transport/BLE.cpp:215
#1  0x00007ffff61f88fa in chip::Ble::BLEEndPoint::DoCloseCallback(unsigned char, unsigned char, int) (this=0x7ffff634cf88 <chip::Ble::BleEndPointPool::Get(unsigned long) const::sEndPointPool>, state=1 '\001', flags=2 '\002', err=6012)
    at ../../src/ble/BLEEndPoint.cpp:439
#2  0x00007ffff61f876d in chip::Ble::BLEEndPoint::FinalizeClose(unsigned char, unsigned char, int) (this=0x7ffff634cf88 <chip::Ble::BleEndPointPool::Get(unsigned long) const::sEndPointPool>, oldState=1 '\001', flags=2 '\002', err=6012)
    at ../../src/ble/BLEEndPoint.cpp:384
#3  0x00007ffff61f7c5a in chip::Ble::BLEEndPoint::DoClose(unsigned char, int) (this=0x7ffff634cf88 <chip::Ble::BleEndPointPool::Get(unsigned long) const::sEndPointPool>, flags=2 '\002', err=6012) at ../../src/ble/BLEEndPoint.cpp:349
#4  0x00007ffff61f7614 in chip::Ble::BleLayer::HandleConnectionError(void*, int) (this=0x7ffff634ceb8 <chip::DeviceLayer::Internal::BLEManagerImpl::sInstance+8>, connObj=0x7fffec007f20, err=6012) at ../../src/ble/BleLayer.cpp:692
#5  0x00007ffff61f47af in chip::DeviceLayer::Internal::BLEManagerImpl::_OnPlatformEvent(chip::DeviceLayer::ChipDeviceEvent const*) (this=0x7ffff634ceb0 <chip::DeviceLayer::Internal::BLEManagerImpl::sInstance>, event=0xb4fe30)
    at ../../src/platform/Linux/BLEManagerImpl.cpp:228
#6  0x00007ffff6202de0 in chip::DeviceLayer::Internal::BLEManager::OnPlatformEvent(chip::DeviceLayer::ChipDeviceEvent const*) (this=0x7ffff634ceb0 <chip::DeviceLayer::Internal::BLEManagerImpl::sInstance>, event=0xb4fe30)
    at ../../src/include/platform/internal/BLEManager.h:171
#7  0x00007ffff6202d01 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToDeviceLayer(chip::DeviceLayer::ChipDeviceEvent const*)
    (this=0x7ffff634d090 <chip::DeviceLayer::PlatformManagerImpl::sInstance>, event=0xb4fe30) at ../../src/include/platform/internal/GenericPlatformManagerImpl.cpp:243
#8  0x00007ffff6202c58 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*)
    (this=0x7ffff634d090 <chip::DeviceLayer::PlatformManagerImpl::sInstance>, event=0xb4fe30) at ../../src/include/platform/internal/GenericPlatformManagerImpl.cpp:199
#9  0x00007ffff6203990 in chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) (this=0x7ffff634d090 <chip::DeviceLayer::PlatformManagerImpl::sInstance>, event=0xb4fe30)
    at ../../src/include/platform/PlatformManager.h:234
#10 0x00007ffff6203931 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::ProcessDeviceEvents() (this=0x7ffff634d090 <chip::DeviceLayer::PlatformManagerImpl::sInstance>)
    at ../../src/include/platform/internal/GenericPlatformManagerImpl_POSIX.cpp:110
#11 0x00007ffff62035fe in chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::SysProcess() (this=0x7ffff634d090 <chip::DeviceLayer::PlatformManagerImpl::sInstance>)
    at ../../src/include/platform/internal/GenericPlatformManagerImpl_POSIX.cpp:180
#12 0x00007ffff62031c6 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() (this=0x7ffff634d090 <chip::DeviceLayer::PlatformManagerImpl::sInstance>)
    at ../../src/include/platform/internal/GenericPlatformManagerImpl_POSIX.cpp:194
#13 0x00007ffff62039f8 in chip::DeviceLayer::PlatformManager::RunEventLoop() (this=0x7ffff634d090 <chip::DeviceLayer::PlatformManagerImpl::sInstance>) at ../../src/include/platform/PlatformManager.h:204
#14 0x00007ffff62037f8 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) (arg=0x7ffff634d090 <chip::DeviceLayer::PlatformManagerImpl::sInstance>)
    at ../../src/include/platform/internal/GenericPlatformManagerImpl_POSIX.cpp:205
#15 0x00007ffff7f81ea7 in start_thread (arg=<optimized out>) at pthread_create.c:477
#16 0x00007ffff7d14def in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

```

mspang, 2021-02-23:
This seems to be a use-after-free of RendezvousSession

```
==1137097== Thread 8:
==1137097== Invalid read of size 8
==1137097==    at 0x75F0E57: chip::RendezvousSession::UpdateState(chip::RendezvousSession::State, int) (RendezvousSession.cpp:248)
==1137097==    by 0x75F13C7: chip::RendezvousSession::OnRendezvousError(int) (RendezvousSession.cpp:239)
==1137097==    by 0x75F7772: chip::Transport::BLE::OnBleEndPointConnectionComplete(chip::Ble::BLEEndPoint*, int) (BLE.cpp:215)
==1137097==    by 0x75B48F9: chip::Ble::BLEEndPoint::DoCloseCallback(unsigned char, unsigned char, int) (BLEEndPoint.cpp:439)
==1137097==    by 0x75B476C: chip::Ble::BLEEndPoint::FinalizeClose(unsigned char, unsigned char, int) (BLEEndPoint.cpp:384)
==1137097==    by 0x75B3C59: chip::Ble::BLEEndPoint::DoClose(unsigned char, int) (BLEEndPoint.cpp:349)
==1137097==    by 0x75B3613: chip::Ble::BleLayer::HandleConnectionError(void*, int) (BleLayer.cpp:692)
==1137097==    by 0x75B07AE: chip::DeviceLayer::Internal::BLEManagerImpl::_OnPlatformEvent(chip::DeviceLayer::ChipDeviceEvent const*) (BLEManagerImpl.cpp:228)
==1137097==    by 0x75BEDDF: chip::DeviceLayer::Internal::BLEManager::OnPlatformEvent(chip::DeviceLayer::ChipDeviceEvent const*) (BLEManager.h:171)
==1137097==    by 0x75BED00: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToDeviceLayer(chip::DeviceLayer::ChipDeviceEvent const*) (GenericPlatformManagerImpl.cpp:243)
==1137097==    by 0x75BEC57: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) (GenericPlatformManagerImpl.cpp:199)
==1137097==    by 0x75BF98F: chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) (PlatformManager.h:234)
==1137097==  Address 0x6854dc0 is 48 bytes inside a block of size 2,560 free'd
==1137097==    at 0x48399AB: free (vg_replace_malloc.c:538)
==1137097==    by 0x7598B16: chip::Platform::MemoryFree(void*) (CHIPMem-Malloc.cpp:132)
==1137097==    by 0x758DB34: void chip::Platform::Delete<chip::RendezvousSession>(chip::RendezvousSession*) (CHIPMem.h:182)
==1137097==    by 0x758EA14: chip::Controller::DeviceCommissioner::RendezvousCleanup(int) (CHIPDeviceController.cpp:735)
==1137097==    by 0x758EABA: chip::Controller::DeviceCommissioner::OnRendezvousError(int) (CHIPDeviceController.cpp:760)
==1137097==    by 0x75F13B9: chip::RendezvousSession::OnRendezvousError(int) (RendezvousSession.cpp:237)
==1137097==    by 0x75F7772: chip::Transport::BLE::OnBleEndPointConnectionComplete(chip::Ble::BLEEndPoint*, int) (BLE.cpp:215)
==1137097==    by 0x75B48F9: chip::Ble::BLEEndPoint::DoCloseCallback(unsigned char, unsigned char, int) (BLEEndPoint.cpp:439)
==1137097==    by 0x75B476C: chip::Ble::BLEEndPoint::FinalizeClose(unsigned char, unsigned char, int) (BLEEndPoint.cpp:384)
==1137097==    by 0x75B3C59: chip::Ble::BLEEndPoint::DoClose(unsigned char, int) (BLEEndPoint.cpp:349)
==1137097==    by 0x75B3613: chip::Ble::BleLayer::HandleConnectionError(void*, int) (BleLayer.cpp:692)
==1137097==    by 0x75B07AE: chip::DeviceLayer::Internal::BLEManagerImpl::_OnPlatformEvent(chip::DeviceLayer::ChipDeviceEvent const*) (BLEManagerImpl.cpp:228)
==1137097==  Block was alloc'd at
==1137097==    at 0x483877F: malloc (vg_replace_malloc.c:307)
==1137097==    by 0x75989B8: chip::Platform::MemoryAlloc(unsigned long, bool) (CHIPMem-Malloc.cpp:112)
==1137097==    by 0x7598932: chip::Platform::MemoryAlloc(unsigned long) (CHIPMem-Malloc.cpp:106)
==1137097==    by 0x758E275: chip::RendezvousSession* chip::Platform::New<chip::RendezvousSession, chip::Controller::DeviceCommissioner*>(chip::Controller::DeviceCommissioner*&&) (CHIPMem.h:168)
==1137097==    by 0x758DDF5: chip::Controller::DeviceCommissioner::PairDevice(unsigned long, chip::RendezvousParameters&, unsigned short, unsigned int) (CHIPDeviceController.cpp:606)
==1137097==    by 0x75210E6: pychip_DeviceController_ConnectBLE (ChipDeviceController-ScriptBinding.cpp:167)
==1137097==    by 0x5919D1C: ??? (in /usr/lib/x86_64-linux-gnu/libffi.so.7.1.0)
==1137097==    by 0x5919288: ??? (in /usr/lib/x86_64-linux-gnu/libffi.so.7.1.0)
==1137097==    by 0x4861569: _ctypes_callproc (in /usr/lib/python3.8/lib-dynload/_ctypes.cpython-38-x86_64-linux-gnu.so)
==1137097==    by 0x4860E2B: ??? (in /usr/lib/python3.8/lib-dynload/_ctypes.cpython-38-x86_64-linux-gnu.so)
==1137097==    by 0x510602: _PyObject_MakeTpCall (in /usr/bin/python3.8)
==1137097==    by 0x50ABF1: _PyEval_EvalFrameDefault (in /usr/bin/python3.8)

```

vivien-apple, 2021-02-25:
Looks like a duplicate of #4356. I guess it answers my question there. It does still happens...

closed 2021-02-25 by vivien-apple