project-chip/connectedhomeip #4984: mDNS needs to properly track provisioning state including operational id

mspang, 2021-02-23:
ESP32 advertises with incorrect service type & operational id

Repro:
- Apply #4976
- Apply #4974
- Flash esp32 & factory reset

```
/out/python_env/bin/chip-device-ctrl
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233

CHIP:DL: CHIP task running
Chip Device Controller Shell

chip-device-ctrl > set-pairing-wifi-credential <ssid> <password>
WiFi credential set
chip-device-ctrl > connect -ble 3840 12345678 9000
Device is assigned with nodeid = 9000
CHIP:DL: TRACE: Bus acquired for name C-eea0
CHIP:DL: TRACE: Bluez mainloop starting BluezMainLoop
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: Long dispatch time: 1000 ms
CHIP:DL: StartDiscovery complete
CHIP:DL: TRACE: Device 77:AE:91:A5:3E:6D Service data: {'0000fea0-0000-1000-8000-00805f9b34fb': <[byte 0x03, 0xfa, 0x8f, 0xca, 0x63, 0xd7, 0xac, 0x70, 0x20, 0x20, 0x20, 0x3e, 0xff]>}
CHIP:DL: TRACE: Device 98:F4:AB:6C:1E:3A Service data: {'0000feaf-0000-1000-8000-00805f9b34fb': <[byte 0x00, 0x00, 0x0f, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: TRACE: Found CHIP BLE Device: 3840
CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.Failed: Software caused connection abort
CHIP:DL: TRACE: Device 98:F4:AB:6C:1E:3A Service data: {'0000feaf-0000-1000-8000-00805f9b34fb': <[byte 0x00, 0x00, 0x0f, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: TRACE: Found CHIP BLE Device: 3840
CHIP:DL: ConnectDevice complete
CHIP:DL: Char1 /org/bluez/hci0/dev_98_F4_AB_6C_1E_3A/service000a
CHIP:DL: Char1 /org/bluez/hci0/dev_98_F4_AB_6C_1E_3A/service000a
CHIP:DL: Char1 /org/bluez/hci0/dev_98_F4_AB_6C_1E_3A/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_98_F4_AB_6C_1E_3A/service000a
CHIP:DL: Char1 /org/bluez/hci0/dev_98_F4_AB_6C_1E_3A/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_98_F4_AB_6C_1E_3A/service000a
CHIP:DL: Char1 /org/bluez/hci0/dev_98_F4_AB_6C_1E_3A/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_98_F4_AB_6C_1E_3A/service000a
CHIP:DL: Char1 /org/bluez/hci0/dev_98_F4_AB_6C_1E_3A/service000a
CHIP:DL: Char1 /org/bluez/hci0/dev_98_F4_AB_6C_1E_3A/service000a
CHIP:DL: Char1 /org/bluez/hci0/dev_98_F4_AB_6C_1E_3A/service000a
CHIP:DL: Char1 /org/bluez/hci0/dev_98_F4_AB_6C_1E_3A/service000a
CHIP:DL: New BLE connection 0x7ff388016240, device 98:F4:AB:6C:1E:3A, path /org/bluez/hci0/dev_98_F4_AB_6C_1E_3A
CHIP:DL: HandlePlatformSpecificBLEEvent 16385
CHIP:DL: HandlePlatformSpecificBLEEvent 16386
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:BLE: subscribe complete, ep = 0x7ff3944bc340
CHIP:DL: Indication received, conn = 0x7ff388016240
CHIP:DL: HandlePlatformSpecificBLEEvent 16388
CHIP:BLE: peripheral chose BTP version 3; central expected between 2 and 3
CHIP:BLE: using BTP fragment sizes rx 20 / tx 128.
CHIP:BLE: local and remote recv window size = 3
CHIP:BLE: Sent PBKDF param request
CHIP:DL: HandlePlatformSpecificBLEEvent 16386
CHIP:DL: Indication received, conn = 0x7ff388016240
CHIP:DL: HandlePlatformSpecificBLEEvent 16388
CHIP:DL: Indication received, conn = 0x7ff388016240
CHIP:DL: HandlePlatformSpecificBLEEvent 16388
CHIP:DL: HandlePlatformSpecificBLEEvent 16386
CHIP:DL: Indication received, conn = 0x7ff388016240
CHIP:DL: HandlePlatformSpecificBLEEvent 16388
CHIP:DL: Indication received, conn = 0x7ff388016240
CHIP:DL: HandlePlatformSpecificBLEEvent 16388
CHIP:DL: HandlePlatformSpecificBLEEvent 16386
CHIP:DL: Indication received, conn = 0x7ff388016240
CHIP:DL: HandlePlatformSpecificBLEEvent 16388
CHIP:BLE: Received PBKDF param response
CHIP:BLE: Sent spake2p msg1
CHIP:DL: HandlePlatformSpecificBLEEvent 16386
[... 475 lines cut ...]
=   eno1 IPv4 0-BC5C01                                      _chip._tcp           local
   hostname = [0-BC5C01.local]
   address = [10.1.0.113]
   port = [11097]
   txt = ["="]

```

This is the right kind of advertisement but wrong operational id. For messaging, the right id is used:

```
I (118356) chip[IN]: Sending msg from 9000 to 112233
I (118366) chip[IN]: Sending secure msg on generic transport
I (118376) chip[IN]: Secure msg send status 0
I (118376) chip[ZCL]: T   0:TX (0x3f40c2f6) Ucast 0x00x3f400c1f
I (118386) chip[ZCL]: TX buffer: [
I (118386) chip[ZCL]: 08 00 0B 02 00 
I (118396) chip[ZCL]: ]
V (118396) chip[ZCL]: Data model processing success!
E (118396) chip[DL]: Long dispatch time: 162 ms
V (119356) chip[DIS]: MinMdns received a query.
V (119366) chip[DIS]: QUERY IN/TXT: 0-bc5c01._chip._tcp.local.
I (119366) chip[DIS]: Broadcasting mDns reply

```

mspang, 2021-02-23:
@pan-apple @andy31415

andy31415, 2021-02-23:
This will depend on the persistence bit for keys - once we have that, we will be able to advertise operationally.

mspang, 2021-02-23:
Sure. That PR was patched in for this test. The device remembers the pairing over a reboot, what's broken is the mDNS advertisement.

woody-apple, 2022-01-25:
Is this resolved? Please reopen if not.

closed 2022-01-25 by woody-apple