project-chip/connectedhomeip #4990: raspi blues central implementation fails to pair

cjandhyala, 2021-02-24:
$ sudo ./out/host/chip-tool pairing ble XYZ ABC 12345678 3840

CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:DL: writing settings to file (/tmp/chip_factory.ini.tmp)
CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)
CHIP:DL: writing settings to file (/tmp/chip_config.ini.tmp)
CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
CHIP:DL: writing settings to file (/tmp/chip_counters.ini.tmp)
CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233

CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:DL: CHIP task running
CHIP:DL: TRACE: Bus acquired for name C-0e73
CHIP:DL: TRACE: Bluez mainloop starting BluezMainLoop
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: Long dispatch time: 1000 ms
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0010
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0010
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0010
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: Char1 /org/bluez/hci0/dev_DC_A6_32_8D_70_74/service0013
CHIP:DL: New BLE connection 0xffff94003aa0, device DC:A6:32:8D:70:74, path /org/bluez/hci0/dev_DC_A6_32_8D_70_74
CHIP:DL: Device already discovered
CHIP:DL: HandlePlatformSpecificBLEEvent 16385
CHIP:DL: FAIL: BluezSendWriteRequest : GDBus.Error:org.bluez.Error.InvalidArguments: Invalid Length
CHIP:BLE: connect handshake timed out, closing ep 0xaaaae802cbe0
CHIP:TOO: Pairing Failure: Ble Error 6021 (0x00001785): Connect handshake timed out
CHIP:BLE: Auto-closing end point's BLE connection.
CHIP:DL: Closing BLE GATT connection (con 0xffff94003aa0)
CHIP:CTL: Shutting down the commissioner
CHIP:CTL: Shutting down the controller
CHIP:DL: BluezDisconnect peer=DC:A6:32:8D:70:74

yunhanw-google, 2021-02-24:
@Damian-Nordic  could you take a look at this? this happens pretty frequent in test setup. chip-device-controller in macbook can make ble connection pretty stable against peripheral device

cjandhyala, 2021-04-06:
Not seeing the issue any more. closing it.

closed 2021-04-06 by cjandhyala