project-chip/connectedhomeip #4994: Align QR code generation base-41 encoding with the spec

bzbarsky-apple, 2021-02-24:
#### Problem
Right now the implementation of QR code generation has an optimization as follows: If the number of bytes to be encoded is even and the last 16-bit value being encoded falls in the range [0x0100, 41*41], then the 16-bit value is encoded in two characters, not 3.  The `41*41` constraint ensures that it fits in two characters; the `0x0100` constraint ensures that these two characters can be disambiguated from a 2-character-encoded single trailing byte.

The problem is that the spec does not have this optimization.  

 #### Proposed Solution
We should align with the spec and just always use three characters for the last 16-bit value when the number of bytes to be encoded is even.

@leorozendaal 

I am not sure what milestone we should use here; seems like the sooner we do this the better, since it involves redoing all QR codes....

@shana-apple

kkasperczyk-no, 2022-05-05:
@bzbarsky-apple,  well I made this change over year ago, so I don't remember all details, but looking at the commit diff and especially this comment: https://github.com/project-chip/connectedhomeip/pull/6132/files#diff-c1d21e4092a28a6f0d7091dfba989a5e90dcbb6dd6f544ef25eaa1e4a83d18f7R132 I would say yes, the optimization was likely removed and any other out-of-spec optimization should not be used.

bzbarsky-apple, 2022-05-05:
@kkasperczyk-no Thank you for double-checking!

closed 2022-05-05 by bzbarsky-apple