project-chip/connectedhomeip #5063: nrfconnect - Unable to declare CPP/CXX flags with quotes

jrhees-cae, 2021-02-28:
Declaring CPP/CXX flags with quotations (escaped or not) causes a build error in `nrfconnect/chip-module/CMakeLists.txt`

Specifically, line 91:
`string(APPEND CHIP_GN_ARGS "${ARG} = filter_exclude(string_split(\"${CFLAGS}\"), ${CFLAG_EXCLUDES})\n")`

For example, adding the following to the application's CMakeLists.txt:
`set(EXTRA_CPPFLAGS '-DSEGGER_RTT_SECTION=".bss"')`
will cause the following error:

```
ERROR at build arg file (use "gn args <out_dir>" to edit):2:47: May only use "." for identifiers.
target_cflags_c = filter_exclude(string_split("-isystem/var/ncs/nrf/drivers/mpsl/clock_control -isystem/var/ncs/zephyr/include -isystem/var/chip/examples/lighting-app/nrfconnect/build/zephyr/include/generated -isystem/var/ncs/zephyr/soc/arm/nordic_nrf/nrf52 -isystem/var/ncs/zephyr/lib/libc/newlib/include -isystem/var/ncs/zephyr/lib/fnmatch/. -isystem/var/ncs/zephyr/subsys/net/lib/sockets/. -isystem/var/ncs/zephyr/subsys/net/lib/openthread/platform/. -isystem/var/ncs/zephyr/subsys/settings/include -isystem/var/ncs/zephyr/subsys/bluetooth -isystem/var/ncs/nrf/include -isystem/var/ncs/nrf/lib/multithreading_lock/. -isystem/var/ncs/nrf/subsys/bluetooth/controller/. -isystem/var/ncs/nrfxlib/nfc/include -isystem/var/ncs/nrfxlib/mpsl/include -isystem/var/ncs/nrfxlib/crypto/nrf_cc310_mbedcrypto/include/mbedtls -isystem/var/ncs/nrfxlib/crypto/nrf_cc310_mbedcrypto/include -isystem/var/chip/examples/lighting-app/nrfconnect/build/modules/nrfxlib/nrf_security/include -isystem/var/chip/examples/lighting-app/nrfconnect/build/modules/nrfxlib/nrf_security/include/mbedtls_generated/ -isystem/var/ncs/mbedtls/include -isystem/var/ncs/mbedtls/include/mbedtls -isystem/var/ncs/nrfxlib/softdevice_controller/include -isystem/var/ncs/modules/hal/cmsis/CMSIS/Core/Include -isystem/var/ncs/modules/hal/nordic/drivers/nrf_radio_802154/. -isystem/var/ncs/modules/hal/nordic/nrfx -isystem/var/ncs/modules/hal/nordic/nrfx/drivers/include -isystem/var/ncs/modules/hal/nordic/nrfx/mdk -isystem/var/ncs/modules/hal/nordic/. -isystem/var/ncs/modules/lib/openthread/zephyr/../include -isystem/var/ncs/modules/lib/openthread/zephyr/../examples/platforms -isystem/var/ncs/modules/debug/segger/rtt -isystem/var/ncs/modules/lib/tinycbor/zephyr/../include -isystem/var/ncs/modules/crypto/tinycrypt/lib/include -isystem/var/ncs/zephyr/../nrfxlib/crypto/nrf_cc310_platform/include -isystem/var/ncs/modules/lib/openthread/zephyr/../include -isystem/var/ncs/modules/lib/openthread/zephyr/../examples/platforms -DKERNEL -D__ZEPHYR__=1 -D_FORTIFY_SOURCE=2 -DBUILD_VERSION=v2.4.0-ncs1 -D__LINUX_ERRNO_EXTENSIONS__ -DUSE_PARTITION_MANAGER=1 -D__PROGRAM_START -DNRF_802154_IRQ_PRIORITY=1 -DNRF_802154_INTERNAL_RADIO_IRQ_HANDLING=0 -DNRF_802154_USE_RAW_API=1 -DNRF_802154_PENDING_SHORT_ADDRESSES=16 -DNRF_802154_PENDING_EXTENDED_ADDRESSES=16 -DNRF_802154_RX_BUFFERS=8 -DNRF_802154_CCA_MODE_DEFAULT=NRF_RADIO_CCA_MODE_ED -DNRF_802154_CCA_CORR_LIMIT_DEFAULT=2 -DNRF_802154_CCA_CORR_THRESHOLD_DEFAULT=45 -DNRF_802154_CCA_ED_THRESHOLD_DEFAULT=45 -DNRF_802154_CSMA_CA_ENABLED=1 -DNRF_802154_TX_STARTED_NOTIFY_ENABLED=1 -DNRF_802154_FRAME_TIMESTAMP_ENABLED=1 -DNRF_802154_DELAYED_TRX_ENABLED=1 -DNRF_802154_ACK_TIMEOUT_ENABLED=1 -DNRF52840_XXAA  -Os -imacros /var/chip/examples/lighting-app/nrfconnect/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m4 -mthumb -mabi=aapcs -imacros /var/ncs/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -DSEGGER_RTT_SECTION=".bss" -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/var/chip/examples/lighting-app/nrfconnect=CMAKE_SOURCE_DIR -fmacro-prefix-map=/var/ncs/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/var/ncs=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs --specs=nosys.specs"), ["-fno-asynchronous-unwind-tables", "-fno-common", "-fno-defer-pop", "-fno-reorder-functions", "-ffunction-sections", "-fdata-sections", "-g*", "-O*", "-W*", ])
                                              ^---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
The thing on the left hand side of the dot must be an identifier
and not an expression. If you need this, you'll have to assign the
value to a temporary first. Sorry.

```

jrhees-cae, 2021-02-28:
@Damian-Nordic found this issue when attempting to add a #define for SEGGER_RTT_SECTION, which requires that the defined value be enclosed in quotes.
Wondering if there is a workaround...

jrhees-cae, 2021-03-01:
FWIW, the following change seems like a possible workaround, but I suspect it may not cover all cases...

```
diff --git a/config/nrfconnect/chip-module/CMakeLists.txt b/config/nrfconnect/chip-module/CMakeLists.txt
index 91dfe155..d50548ce 100644
--- a/config/nrfconnect/chip-module/CMakeLists.txt
+++ b/config/nrfconnect/chip-module/CMakeLists.txt
@@ -85,7 +85,8 @@ macro(chip_gn_arg_cflags ARG CFLAGS)
         string(APPEND CFLAG_EXCLUDES "\"${cflag}\", ")
     endforeach()
     string(APPEND CFLAG_EXCLUDES "]")
-    string(APPEND CHIP_GN_ARGS "${ARG} = filter_exclude(string_split(\"${CFLAGS}\"), ${CFLAG_EXCLUDES})\n")
+    string(REGEX REPLACE "([][\"])" "\\\\\\1" gn_args_escaped "${CFLAGS}")
+    string(APPEND CHIP_GN_ARGS "${ARG} = filter_exclude(string_split(\"${gn_args_escaped}\"), ${CFLAG_EXCLUDES})\n")
 endmacro()
 
 # ==============================================================================
```

mspang, 2021-03-01:
Yeah, we need to shell escape the args here. See https://gn.googlesource.com/gn/+/master/docs/language.md#strings for grammar help.

jrhees-cae, 2021-03-01:
@mspang Are you suggesting there's a way to shell-escape these quotes at the source? I originally tried that (e.g. with `set(EXTRA_CPPFLAGS '-DSEGGER_RTT_SECTION=\".bss\"')`. I even tried double-escaping (\\\") but the escapes are all stripped off by the time it reaches the CMakeLiists.txt processing later. Hence the snippet of code above that re-escapes the quotes.

Damian-Nordic, 2021-03-01:
@jrhees-cae 
I just checked that something like `set(CHIP_CFLAGS -DABC=\\\\"abc\\\\")` works. When you use such syntax, CMake stores `-DABC=\\"abc\\"` into the variable, which later on is resolved to `-DABC=\"abc\"` in the args.gni file. Of course, we will do the escaping in the CMake module itself, but for now you may use it as a workaround, I guess.

jrhees-cae, 2021-03-01:
@Damian-Nordic unfortunately that double-escape method doesn't work, as it creates problems with the C Preprocessor where the #define is actually used. Part of the problem is that the SEGGER_RTT_SECTION #define that I've used in the example needs a quoted string without any escapes, and will not accept anything else.

```
<command-line>: warning: missing terminating " character
<command-line>: error: stray '\' in program
/var/ncs/modules/debug/segger/rtt/SEGGER_RTT.c:197:75: note: in definition of macro 'SEGGER_RTT_PUT_SECTION'
  197 |     #define SEGGER_RTT_PUT_SECTION(Var, Section) __attribute__ ((section (Section))) Var
      |                                                                           ^~~~~~~
```

Damian-Nordic, 2021-03-01:
Ah, right. So I guess we just need to apply your fix.

PR #5100 (2021-03-02): [nrfconnect] Escape GN arguments coming from CMake
 #### Problem
CMake flags that are passed to GN and contain GN special characters like `"`, `\` and `$` are not escaped and break the build.

 #### Summary of Changes
Provide a dedicated Python script since some of the GN arguments are calculated using CMake generator
expressions which are only evaluated in the build phase, so the escaping can't be fully done in CMake alone.

 Fixes #5063
closed 2021-03-02 by bzbarsky-apple