project-chip/connectedhomeip #5097: Data received on an unknown connection (0). Dropping it!!

sebcerdan, 2021-03-02:
Hello,

When using Chiptool to drive lighting-app i have the following log on RTT viewer:
L] Thread packet RCVD: UDP, len 108
0> [DL] src 2002::1, port 11098
0> [DL] dest 2001:DB8::C556:8EDF:B33B:FEF8, port 11097
0> [IN] Data received on an unknown connection (0). Dropping it!!
0> [SVR] Packet received error: Error 4000127 (0x003D097F)
0> <efr32 > Device is commissioned to a Thread network.
0> [DL] Thread packet RCVD: UDP, len 110
0> [DL] src 2002::1, port 11098
0> [DL] dest 2001:DB8::C556:8EDF:B33B:FEF8, port 11097
0> [IN] Data received on an unknown connection (0). Dropping it!!
0> [SVR] Packet received error: Error 4000127 (0x003D097F)

I use the two following command on chiptool:

./chip-tool pairing bypass 2001:db8:0:0:c556:8edf:b33b:fef8 11097

CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233

CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:TOO: Pairing Success
CHIP:CTL: Shutting down the commissioner
CHIP:CTL: Shutting down the controller
CHIP:DL: CHIP task running

./chip-tool onoff read on-off 1
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233

CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:TOO: Sending cluster (0x0006) command (0x00) on endpoint 1
CHIP:ZCL: Successfully encoded 11 bytes
CHIP:IN: New pairing for device 12344321, key 0!!
CHIP:IN: Secure message was encrypted: Msg ID 0
CHIP:IN: Sending msg from 112233 to 12344321
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status 0
CHIP:CTL: SendMessage returned 0
CHIP:DL: CHIP task running
CHIP:TOO: No response from device
CHIP:CTL: Shutting down the commissioner
CHIP:CTL: Shutting down the controller
CHIP:TOO: Run command failure: Error 4172 (0x0000104C)

I have a brd4161a, a raspberry acting as OTBR with brd4166a as RCP and a Ubuntu VM running Chiptool.
I mounted this setup yesterday with last GIT version for all the SW.

What is wrong here?

QueZinger00621, 2021-03-04:
Same thing happened. BTW, how can we know the endpoint-id without any hint?

silabs-NikhilS, 2021-03-05:
I had similar issue  yesterday. But my issue was that I was commissioning the device on thread network via the commissioning-cli tool on the BR and not pairing it via the chip-tool . But when I provisioned the CHIP device via the chip-tool android app by scanning the QR code on LCD of the WSTK board, I no longer see the error and can send the on-off commands successfully from the android app. I haven't tried the chip-tool linux app as I had some issues building it.

One tip when using the android app - you have to enable the location service access to the android app to get the pairing done successfully. Hope it helps!

362dyx, 2021-03-11:
I have the same problem. Have you solved it

sebcerdan, 2021-03-11:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Use python controller instead of chip tool

362dyx, 2021-03-11:
Do you know the reason?Why doesn't Chiptool work

hainahouwei, 2021-05-08:
> 
> Do you know the reason?Why doesn't Chiptool work

I used python controller on linux virtual machine ,and followed the file https://github.com/project-chip/connectedhomeip/blob/master/src/controller/python/README.md.

```
sudo chip-device-ctrl
```
```
set-pairing-thread-credential <channel> <pan id[HEX]> <master_key>
```

```
connect -ble <discriminator> <setup pin code> [<nodeid>]
```
However it can't conneted to the device with cmd " connect -ble <discriminator> <setup pin code> [<nodeid>]" .

Are there anyone facing the same problem?
Must I run python controller  on Pi?

log on linux:
```
chip-device-ctrl > connect -ble 3840 012345678 
Device is assigned with nodeid = 725988
CHIP:BLE: BLE scanning through known devices.
CHIP:BLE: New device scanned: 00:0B:57:D8:F0:D7
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Failed to stop discovery GDBus.Error:org.bluez.Error.Failed: No discovery started
CHIP:BLE: Scan complete notification without an active scan.
CHIP:BLE: Device 4E:4C:EE:0C:A8:E8 does not look like a CHIP device.
CHIP:BLE: Device 54:A0:DC:1B:17:5C does not look like a CHIP device.
CHIP:BLE: Device A4:50:46:0B:5D:78 does not look like a CHIP device.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: Failed to start discovery: Operation was cancelled
CHIP:BLE: Scan complete notification without an active scan.
CHIP:DL: ConnectDevice complete
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0015
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0015
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0015
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0001
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0001
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0001
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0001
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0001
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0001
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service000e
CHIP:DL: Char1 /org/bluez/hci0/dev_00_0B_57_D8_F0_D7/service0018
[... 177 lines cut ...]
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0x7f07b0046120
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0x7f07b0046120
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0x7f07b0046120
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0x7f07b0046120
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0x7f07b0046120
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Bluez disconnected
CHIP:DL: Bluez notify CHIPoBluez connection disconnected
CHIP:CTL: Remote device failed in SPAKE2+ handshake

CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
Failed to establish secure session to device: 6012
Device temporary node id (**this does not match spec**): 725988
```

woody-apple, 2022-01-25:
Is this resolved? Please reopen if not.

closed 2022-01-25 by woody-apple