project-chip/connectedhomeip #5140: CHIP-tool SEGFAULTS on master when trying to provision ESP32

andy31415, 2021-03-03:
#### Problem

Valgrind sees a 'use after free' when we encounter a pairing error:

```
CHIP:TOO: Secure Pairing Failed                                                                                                                                                                                                      
CHIP:TOO: Pairing Failure: Error 4047 (0x00000FCF)                                                                                                                                                                                   
==949307== Thread 6:                                                                                                                                                                                                                 
==949307== Invalid read of size 8                                                                                                                                                                                                    
==949307==    at 0x1CAFB6: chip::RendezvousSession::UpdateState(chip::RendezvousSession::State, int) (RendezvousSession.cpp:271)                                                                                                     
==949307==    by 0x1CAF66: chip::RendezvousSession::OnRendezvousError(int) (RendezvousSession.cpp:262)                                                                                                                               
==949307==    by 0x1CAA7A: chip::RendezvousSession::OnSessionEstablishmentError(int) (RendezvousSession.cpp:163)                                                                                                                     
==949307==    by 0x1C9CE7: chip::PASESession::HandlePeerMessage(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) (PASESession.cpp:828)                                              
==949307==    by 0x1CB469: chip::RendezvousSession::HandlePairingMessage(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) (RendezvousSession.cpp:393)                               
==949307==    by 0x1CB2AF: chip::RendezvousSession::OnRendezvousMessageReceived(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) (RendezvousSession.cpp:365)                        
==949307==    by 0x1CFBE6: chip::Transport::BLE::OnBleEndPointReceive(chip::Ble::BLEEndPoint*, chip::System::PacketBufferHandle) (BLE.cpp:197)                                                                                       
==949307==    by 0x1F6E8F: chip::Ble::BLEEndPoint::Receive(chip::System::PacketBufferHandle) (BLEEndPoint.cpp:1434)                                                                                                                  
==949307==    by 0x1F867A: chip::Ble::BleLayer::HandleIndicationReceived(void*, chip::Ble::ChipBleUUID const*, chip::Ble::ChipBleUUID const*, chip::System::PacketBufferHandle) (BleLayer.cpp:500)                                   
==949307==    by 0x1A8CE9: chip::DeviceLayer::Internal::BLEManagerImpl::HandlePlatformSpecificBLEEvent(chip::DeviceLayer::ChipDeviceEvent const*) (BLEManagerImpl.cpp:282)                                                           
==949307==    by 0x1A8B2E: chip::DeviceLayer::Internal::BLEManagerImpl::_OnPlatformEvent(chip::DeviceLayer::ChipDeviceEvent const*) (BLEManagerImpl.cpp:254)                                                                         
==949307==    by 0x1AC410: chip::DeviceLayer::Internal::BLEManager::OnPlatformEvent(chip::DeviceLayer::ChipDeviceEvent const*) (BLEManager.h:171)                                                                                    
==949307==  Address 0x5d04550 is 48 bytes inside a block of size 2,560 free'd                                                                                                                                                        
==949307==    at 0x48399AB: free (vg_replace_malloc.c:538)                                                                                                                                                                           
==949307==    by 0x1A5036: chip::Platform::MemoryFree(void*) (CHIPMem-Malloc.cpp:132)                                                                                                                                                
==949307==    by 0x1A025B: void chip::Platform::Delete<chip::RendezvousSession>(chip::RendezvousSession*) (CHIPMem.h:187)                                                                                                            
==949307==    by 0x19FA25: chip::Controller::DeviceCommissioner::RendezvousCleanup(int) (CHIPDeviceController.cpp:738)                                                                                                               
==949307==    by 0x19FADF: chip::Controller::DeviceCommissioner::OnRendezvousError(int) (CHIPDeviceController.cpp:763)                                                                                                               
==949307==    by 0x1CAF52: chip::RendezvousSession::OnRendezvousError(int) (RendezvousSession.cpp:260)                                                                                                                               
==949307==    by 0x1CAA7A: chip::RendezvousSession::OnSessionEstablishmentError(int) (RendezvousSession.cpp:163)                                                                                                                     
==949307==    by 0x1C9CE7: chip::PASESession::HandlePeerMessage(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) (PASESession.cpp:828)                                              
==949307==    by 0x1CB469: chip::RendezvousSession::HandlePairingMessage(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) (RendezvousSession.cpp:393)                               
==949307==    by 0x1CB2AF: chip::RendezvousSession::OnRendezvousMessageReceived(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) (RendezvousSession.cpp:365)                        
==949307==    by 0x1CFBE6: chip::Transport::BLE::OnBleEndPointReceive(chip::Ble::BLEEndPoint*, chip::System::PacketBufferHandle) (BLE.cpp:197)                                                                                       
==949307==    by 0x1F6E8F: chip::Ble::BLEEndPoint::Receive(chip::System::PacketBufferHandle) (BLEEndPoint.cpp:1434)                                                                                                                  
==949307==  Block was alloc'd at                                                                                                                                                                                                     
==949307==    at 0x483877F: malloc (vg_replace_malloc.c:307)                                                                                                                                                                         
==949307==    by 0x1A4FAC: chip::Platform::MemoryAlloc(unsigned long, bool) (CHIPMem-Malloc.cpp:112)                                                                                                                                 
==949307==    by 0x1A4F7D: chip::Platform::MemoryAlloc(unsigned long) (CHIPMem-Malloc.cpp:106)                                                                                                                                       
==949307==    by 0x1A027B: chip::RendezvousSession* chip::Platform::New<chip::RendezvousSession, chip::Controller::DeviceCommissioner*>(chip::Controller::DeviceCommissioner*&&) (CHIPMem.h:168)                                     
==949307==    by 0x19F49D: chip::Controller::DeviceCommissioner::PairDevice(unsigned long, chip::RendezvousParameters&) (CHIPDeviceController.cpp:612)                                                                               
==949307==    by 0x131C43: PairingCommand::Pair(unsigned long, chip::Transport::PeerAddress) (PairingCommand.cpp:74)                                                                                                                 
[... cut ...]
==949307==    by 0x12C61D: Commands::RunCommand(PersistentStorage&, unsigned long, unsigned long, int, char**) (Commands.cpp:118)                                                                                                    
==949307==    by 0x12BC05: Commands::Run(unsigned long, unsigned long, int, char**) (Commands.cpp:49)                                                                                                                                
==949307==    by 0x166A8E: main (main.cpp:41

```

andy31415, 2021-03-03:
Bisected and it seems like 93b6456850f74c49b2e39a94f67e0d90e651e290 is the first commit to fail (#5081)

I suspect that the PR just injects a failure however that should not make us use freed memory, so we seem to absolutely have a ownership/lifetime issue here.

andy31415, 2021-03-03:
Apparently our current salt size (at least for my runs) is 25, which is an error after that PR (min 8, max 16).

I will look into fixing the invalid memory access first and only then fixing the salt size.

andy31415, 2021-03-03:
error is because "SPAKE2P Key Exchange Salt" is longer than the expected max.

Did not figure out the crash yet

andy31415, 2021-03-03:
Apparently #5081 also changed the salt string to be within bounds. For me however, I had a old flash of ESP32, so the device sends a long salt.

To reproduce:
  - flash a pre #5081 build of ESP32
  - run chip-tool to provision

vivien-apple, 2021-03-04:
I think this one, #4965 and #4356 does have the same root cause.

Following the steps to reproduce, it does not crash for me.
But looking at the code those lines seems suspicious:
 https://github.com/project-chip/connectedhomeip/blob/dcb711136e006e59258d07a60dbdaac60e1f2bba/src/transport/RendezvousSession.cpp#L245

So when there is an error, the code calls the delegate method `OnRendezvousError`, then  the commissioner calls `DeviceCommissioner::RendezvousCleanup` which itself does: 
```
    if (mRendezvousSession != nullptr)                                                                                                                                                
    {                                                                                                                                                                                 
        chip::Platform::Delete(mRendezvousSession);                                                                                                                                   
        mRendezvousSession = nullptr;                                                                                                                                                 
    } 
```

But `RendezvousSession` is not done yet, and it tries to continue to do some clean up after it has been free'd...

andy31415, 2021-03-04:
Reproduction steps:

Build a ESP32 that sends a long salt:

```
git checkout 2454d646
# build esp32
./examples/all-clusters-app/esp32/build/chip-all-clusters-app.flash.py --erase
```

followed by trying to provision that using a new chip-tool (this will error out in chip-tool because of invalid key lenght):

```
checkout master
# build chip-tool
./out/chip_tool/chip-tool pairing ble ssid password 12345678 3840
```

PR #5156 (2021-03-04): CHIP-tool SEGFAULTS on master when trying to provision ESP32
 #### Problem

`RendezvousSession` is beeing used after it has been free`d.

Sadly I can not reproduce the crash locally, so this is a tentative to fix it. 
@andreilitvin Any chance you can test it ?

 #### Summary of Changes
 * Update when the delegate is called

 Fixes #5140, #4965 and #4356
closed 2021-03-04 by andy31415