project-chip/connectedhomeip #5147: python controller shuts down without pairing over BT on a nordic DK

cjandhyala, 2021-03-04:
python controller shuts down without pairing over BT on a nordic DK 

$ chip-device-ctrl 
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233

CHIP:DL: CHIP task running
Chip Device Controller Shell

chip-device-ctrl > set-pairing-thread-credential 15 1234 00112233445566778899aabbccddeeff
Thread credential set
chip-device-ctrl > connect -ble 3000 12345678 12344321
Device is assigned with nodeid = 12344321
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:DL: TRACE: Bus acquired for name C-0ea5
CHIP:DL: PlatformBlueZInit init success
CHIP:BLE: BLE scanning through known devices.
CHIP:BLE: Device C2:02:86:8F:C5:29 does not look like a CHIP device.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: Device 5C:61:C4:91:B4:DA does not look like a CHIP device.
CHIP:BLE: Device 5E:4A:3A:C7:28:E6 does not look like a CHIP device.
CHIP:BLE: Device 58:4F:B7:2E:C0:B6 does not look like a CHIP device.
CHIP:BLE: Device 27:B7:A0:15:C4:43 does not look like a CHIP device.
CHIP:BLE: Device 4B:E0:8F:1A:D9:A6 does not look like a CHIP device.
CHIP:BLE: Device F3:33:19:1E:96:07 does not look like a CHIP device.
CHIP:BLE: Device 71:6E:82:CF:A9:A6 does not look like a CHIP device.
CHIP:BLE: Device 08:66:98:91:F5:27 does not look like a CHIP device.
CHIP:BLE: Device 76:87:EB:24:0D:45 does not look like a CHIP device.
CHIP:BLE: Device 08:66:98:91:66:E3 does not look like a CHIP device.
CHIP:BLE: Device 6B:82:87:6B:E2:1D does not look like a CHIP device.
CHIP:BLE: Device 74:72:CF:3C:E3:4E does not look like a CHIP device.
CHIP:BLE: Device 48:80:18:72:53:1E does not look like a CHIP device.
CHIP:BLE: Device 15:93:98:EA:8D:B4 does not look like a CHIP device.
CHIP:BLE: Device 5E:E1:5A:40:D2:7E does not look like a CHIP device.
CHIP:BLE: Device 47:AC:0D:64:6A:75 does not look like a CHIP device.
CHIP:BLE: Device 42:D8:8A:E2:C2:EB does not look like a CHIP device.
CHIP:BLE: Device 2C:0D:FD:CB:AB:0A does not look like a CHIP device.
CHIP:BLE: Device 54:97:80:15:13:58 does not look like a CHIP device.
CHIP:BLE: Device 65:7A:75:8E:FA:3A does not look like a CHIP device.
CHIP:BLE: Device 65:19:47:A0:9C:63 does not look like a CHIP device.
CHIP:BLE: Device 18:28:9B:86:C3:B5 does not look like a CHIP device.
CHIP:BLE: Device 6E:AA:91:7A:77:94 does not look like a CHIP device.
CHIP:BLE: Device 60:CC:75:13:1C:7C does not look like a CHIP device.
CHIP:BLE: Device 24:3C:A9:2E:9B:33 does not look like a CHIP device.
CHIP:BLE: Device 5C:C8:65:19:2D:C3 does not look like a CHIP device.
CHIP:BLE: Device 50:33:D8:C1:F0:27 does not look like a CHIP device.
CHIP:BLE: Device 54:38:B3:A4:A1:5C does not look like a CHIP device.
CHIP:BLE: Device FB:67:AC:AE:05:8A does not look like a CHIP device.
CHIP:BLE: Device 1B:BB:9A:40:7B:D3 does not look like a CHIP device.
CHIP:BLE: Device F8:7D:40:2B:61:38 does not look like a CHIP device.
CHIP:BLE: Device 5C:61:C4:91:B4:DA does not look like a CHIP device.
CHIP:BLE: Device 7A:AA:3C:10:6E:41 does not look like a CHIP device.
CHIP:BLE: Device 08:66:98:91:F5:27 does not look like a CHIP device.
Failed to establish secure session to device: 4050
Device temporary node id (**this does not match spec**): 12344321
chip-device-ctrl > connect -ble 3000 12345678 12344321
Device is assigned with nodeid = 12344321
Error 4011 (0x00000FAB)
chip-device-ctrl > quit
CHIP:CTL: Shutting down the commissioner
CHIP:CTL: Shutting down the controller
CHIP:DL: System Layer shutdown
CHIP:DL: Inet Layer shutdown
CHIP:IN: Async DNS worker thread woke up.
CHIP:IN: Async DNS worker thread exiting.
CHIP:IN: Async DNS worker thread woke up.
CHIP:IN: Async DNS worker thread exiting.
CHIP:DL: BLE layer shutdown

LuDuda, 2021-03-04:
Few questions:
 - Have you pushed button 4 on the DK?
 - I see you use discriminator value of 3000 instead of default 3840, have you changed it in the code?
 - If above does not work, what are you using for controlller - RaspPi, Linux, macOS? For Linux it seems to work quite smoothly.

cjandhyala, 2021-03-09:
Yes Lukas, as discussed with you, Damian in slack, all of those steps followed. However I don't see this issue any more with ToT on master branch , hence closing the issue.

closed 2021-03-09 by cjandhyala