project-chip/connectedhomeip #5179: chip-tool api has been changed?

yuzibo, 2021-03-05:
Hi,  I am very new to CHIP and I am excited about it.

I tested it on two raspaberry pi 4b for lighting-app.
1.  Refer to  the
 [link](https://github.com/projectchip/connectedhomeip/tree/5cf968c502fca1eecdc62ed3801f07252e6b0369/examples/lighting-app/linux), I built the chip-lighting-app on PI-2:

```bash
ubuntu@pi-2:~/git/connectedhomeip/examples/lighting-app/linux$ sudo out/debug/chip-lighting-app --ble-device 1
SetupPINCode: [12345678]
SetupQRCode:  [CH:I34DV*700 0C9SS0]
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 12344321

CHIP:DL: wpa_supplicant: _IsWiFiStationProvisioned: interface not connected
CHIP:BLE: Waiting for PBKDF param request
CHIP:DIS: CHIP minimal mDNS configured as 'Commisioning device'.
CHIP:DIS: CHIP minimal mDNS started advertising.
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Broadcasting mDns reply
CHIP:DIS: Broadcasting mDns reply
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Broadcasting mDns reply
CHIP:DIS: Broadcasting mDns reply
CHIP:SVR: Server Listening...
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:DL: TRACE: Bus acquired for name CHIP-3840
CHIP:DL: CREATE service object at /chipoble/2d19/service
CHIP:DL: Create characteristic object at /chipoble/2d19/service/c1
CHIP:DL: Create characteristic object at /chipoble/2d19/service/c2
CHIP:DL: CHIP BTP C1 /chipoble/2d19/service
CHIP:DL: CHIP BTP C2 /chipoble/2d19/service
CHIP:DL: CHIP_ENABLE_ADDITIONAL_DATA_ADVERTISING is TRUE
CHIP:DL: Create characteristic object at /chipoble/2d19/service/c3
CHIP:DL: rotatingDeviceId: 00007C5F6E176CD40F68685D100A1CF8A98B
CHIP:DL: CHIP BTP C3 /chipoble/2d19/service
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: BluezPeripheralRegisterAppDone done
CHIP:DL: HandlePlatformSpecificBLEEvent 16391
CHIP:DL: Create adv object at /chipoble/2d19/advertising
CHIP:DL: SET service data to {'0xFEAF': <[byte 0x00, 0x00, 0x0f, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: HandlePlatformSpecificBLEEvent 16392
CHIP:DL: CHIPoBLE advertising config complete
CHIP:DL: RegisterAdvertisement complete
CHIP:DL: HandlePlatformSpecificBLEEvent 16393
CHIP:DL: New BLE connection 0xffff78035b70, device B8:27:EB:4F:32:C9, path /org/bluez/hci1/dev_B8_27_EB_4F_32_C9
CHIP:DL: BluezCharacteristicAcquireWrite is called, conn: 0xffff78035b70
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 517
CHIP:DL: FAIL: short read in BluezCharacteristicWriteFD (0)
CHIP:DL: CHIPoBLE subscribe received
CHIP:BLE: no endpoint for sub recvd
CHIP:DL: HandlePlatformSpecificBLEEvent 32777
CHIP:DL: Receive kCHIPoBLEConnectionEstablished
```

And I referred to [link](https://github.com/project-chip/connectedhomeip/tree/master/examples/chip-tool) to compile chip-tool.
When I am pairing ble-device on pi-2 with it, I am confused about:
```bash
ubuntu@ubuntu-1:~/connectedhomeip/examples/chip-tool$ sudo out/debug/chip-tool pairing ble  12345678 3840
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:TOO: InitArgs: Wrong arguments number: 2 instead of 4
Usage:
  out/debug/chip-tool pairing ble ssid password setup-pin-code discriminator
```
Where do ssid and passwd come from? Can it generate an AP with ble-device?
My final goal is very easy, it is to enable communication with each other pi 4b with CHIP arch.
I try it many ways but fail to do it. 
Any ideas are appreciated. Thank you.

yuzibo, 2021-03-05:
Ok, I don't know where ssid and passwd come from, though. I fill them with an AP, which is two pi ap also. It display:
```bash
ubuntu@ubuntu-1:~/connectedhomeip/examples/chip-tool$ sudo out/debug/chip-tool pairing ble AP AP_PASSWD   12345678 3840
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233

CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:DL: CHIP task running
CHIP:DL: TRACE: Bus acquired for name C-1726
CHIP:DL: TRACE: Bluez mainloop starting BluezMainLoop
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: Long dispatch time: 1000 ms
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service0006
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: Char1 /org/bluez/hci0/dev_B8_27_EB_A0_97_2B/service002e
CHIP:DL: New BLE connection 0xffff74003aa0, device B8:27:EB:A0:97:2B, path /org/bluez/hci0/dev_B8_27_EB_A0_97_2B
CHIP:DL: Device already discovered
CHIP:DL: HandlePlatformSpecificBLEEvent 16385
CHIP:DL: HandlePlatformSpecificBLEEvent 16386
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:BLE: subscribe complete, ep = 0xaaaabdffcbe8
CHIP:BLE: connect handshake timed out, closing ep 0xaaaabdffcbe8
CHIP:TOO: Pairing Failure: Ble Error 6021 (0x00001785): Connect handshake timed out
CHIP:CTL: Shutting down the commissioner
CHIP:CTL: Shutting down the controller
ubuntu@ubuntu-1:~/connectedhomeip/examples/chip-tool$

```

jmeg-sfy, 2021-03-09:
Same issue for me here.
SSID and Password are supposed to be BR SoftAp right ?
The main doc page is leading to confusion

yuzibo, 2021-03-09:
Hi, I can test light-app with onoff mode.
In fact, When to compiling example/light-app/chip-lighting-app, you have to enable "chip_bypass_rendezvous=true" options.
Then you do ./chip-tool bypass ip 11097

vivien-apple, 2021-04-02:
Currently `ssid` and `password` are the ssid/password of the wifi network you want the device to connect to. So those are not the ssid/password of the SoftAp of the device itself.

closed 2022-10-18 by stale[bot]