project-chip/connectedhomeip #5191: Followup isues for #4932

bzbarsky-apple, 2021-03-05:
#### Problem
I finally got a chance to review #4932 and ran into various issues.  I'll fix up some of them in https://github.com/project-chip/connectedhomeip/pull/5190 but the remaining ones are (and feel free to split these out into multiple issues as needed):

* [ ] We are storing the `IsPeerGroupMsgIdNotSynchronized` state in the PacketHeader, which seems like as lightly odd place to put it _and_ requires `const_cast`.  We should not be adding `const_cast` to the code.
* [ ] In `ExchangeContext::SendMessage` we are triggering sync for all messages, not just reliable ones.  Per spec we only sync on send if we're sending a reliable message (because then we expect an ack).
* [ ] We have duplicated code for checking "is this a message counter sync message?"  Why do we not just put this on payloadHeader?

 #### Proposed Solution
Resolve the above.

@yufengwangca

yufengwangca, 2021-03-07:
For #1, Now we have three parameters to pass this meta information to messaging layer for incoming packet. packetHeader, payloadHeader, secureHandle. Semantically, PacketHeader is the best option, this is also aligned with Weave. But we need to use const_cast if use PacketHeader. The other two do not need const_cast, maybe use SecureSessionHandle instead?   

For #2, The spec say "Peer message counter synchronization is an essential part of enabling secure CHIP messaging between members of an application group"  and it is only used for group messing right now.

CHIP_ERROR ExchangeContext::SendMessage(uint16_t protocolId, uint8_t msgType, PacketBufferHandle msgBuf,
                                        const SendFlags & sendFlags)
{
   .....
    // If a group message is to be transmitted to a destination node whose message counter is unknown.
    if (ChipKeyId::IsAppGroupKey(state->GetLocalKeyID()) && !state->IsPeerMsgCounterSynced())
    {
         .... Initiate message counter synchronization if no message counter synchronization is in progress.
    }

 ......
}

Need clarify what type of messages need this protocol.

#3, putting into payloadHeader can also solve #1, but this information is clearly not part of payload (Exchange Layer).

bzbarsky-apple, 2021-03-09:
> But we need to use const_cast if use PacketHeader. 

If this place needs to modify the PacketHeader, it should not take it as const.  If we feel the PacketHeader here should be const, we should do the modification at a lower level, where we have a non-const one.

But also, I don't see why PacketHeader is semantically the right thing, if it's actually supposed to just represent the packet header information.  The synchronization state is a property of the session, it seems like, not of the header...  The header can serve as input to the session to decide whether we're synchronized.

> The spec say

Yes, I'm aware of what the spec says.  If you're _sending_ a message, the only reason to do sync is if you expect a response.  The only cases in which the message layer _knows_ it expects a response is if you're sending a unicast message, but using a group key, and the R flag is set, so we expect an ack....

> but this information is clearly not part of payload

How not?  The message type is in the payload header, and our question is "is this a message of a certain type?"

yufengwangca, 2021-03-15:
Problem 1 is addressed in https://github.com/project-chip/connectedhomeip/pull/5368

yufengwangca, 2021-04-29:
Mingjie is currently refactoring the MCSP implementation, no need to spend effort to improve current implementation.

closed 2021-04-29 by yufengwangca