project-chip/connectedhomeip #5196: Crash in chip::Transport::BLE::OnBleConnectionError

sagar-apple, 2021-03-05:
#### Problem
Possibly related https://github.com/project-chip/connectedhomeip/issues/5193

```
Thread 8 name:  Dispatch queue: com.chip.ble.work_queue
Thread 8 Crashed:
0   CHIP                          	0x0000000100e86d04 chip::Transport::BLE::OnBleConnectionError(void*, int) + 56
1   CHIP                          	0x0000000100e68b58 __39-[BleConnection initWithDiscriminator:]_block_invoke + 76
2   CHIP                          	0x0000000100e68b58 __39-[BleConnection initWithDiscriminator:]_block_invoke + 76
3   libdispatch.dylib             	0x0000000189ac8d04 _dispatch_client_callout + 20
4   libdispatch.dylib             	0x0000000189acc36c _dispatch_continuation_pop + 484
5   libdispatch.dylib             	0x0000000189adfe14 _dispatch_source_invoke + 1300
6   libdispatch.dylib             	0x0000000189ad06fc _dispatch_lane_serial_drain + 300
7   libdispatch.dylib             	0x0000000189ad1444 _dispatch_lane_invoke + 412
8   libdispatch.dylib             	0x0000000189adc174 _dispatch_workloop_worker_thread + 768
9   libsystem_pthread.dylib       	0x00000001dc1bf704 _pthread_wqthread + 276
10  libsystem_pthread.dylib       	0x00000001dc1c6744 start_wqthread + 8
```

 #### Proposed Solution
It looks like we don't clean up the ble state correctly. When the OnBleConnectionError is triggered we have likely already released 
the context.

vivien-apple, 2021-03-09:
@sagar-apple Does it still happens with #5219 ?

woody-apple, 2022-01-25:
Is this resolved? Please reopen if not.

closed 2022-01-25 by woody-apple