project-chip/connectedhomeip #5273: Nordic thread accessory can't decrypt CHIP messages sent over UDP

cjandhyala, 2021-03-09:
Nordic thread accessory can't decrypt CHIP messages sent over UDP

Accessory log
-------------

D: 29307 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 29315 [DL]Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 29322 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 29403 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 29410 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 29505 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 29512 [DL]Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 29520 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 29600 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 29608 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 8)
D: 29696 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 29704 [DL]Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 29711 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 29792 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 29800 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 20)
D: 29894 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
I: 31548 [DL]BLE GAP connection terminated (reason 0x13)
I: 31557 [DL]Current number of connections: 0/1
D: 31561 [DL]In DriveBLEState
I: 31572 [DL]CHIPoBLE advertising started
E: 41773 [CR]mbedTLS error: 

E: 44718 [BLE]ack recv timeout, closing ep 0x20006378
D: 44723 [DL]SetAdvertisingEnabled(false)
I: 44727 [SVR]Persisting admin ID 0, next available 1
I: 44964 [SVR]Persisting admin ID successfully
I: 44968 [SVR]OnRendezvousError: Ble Error 6024 (0x00001788): Message fragment acknowlt
I: 44977 [BLE]Releasing end point's BLE connection back to application.
D: 44984 [DL]In DriveBLEState
I: 44987 [DL]CHIPoBLE advertising stopped
E: 49648 [CR]mbedTLS error: 

E: 63034 [CR]mbedTLS error: 

CTRL-A Z for help | 115200 8N1 | NOR | Minicom 2.8 | VT102 | Offline | em0006830147531

Controller side log
------------------
$ chip-device-ctrl 
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 18446743528248704968

CHIP:DL: CHIP task running
Chip Device Controller Shell

chip-device-ctrl > set-pairing-thread-credential 11 abcd 00112233445566778899aabbccddeeff
Thread credential set
chip-device-ctrl > connect -ble 3000 12345678 12344321
Device is assigned with nodeid = 12344321
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:DL: TRACE: Bus acquired for name C-9b95
CHIP:DL: PlatformBlueZInit init success
CHIP:BLE: BLE scanning through known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: Device FB:67:AC:AE:05:8A does not look like a CHIP device.
CHIP:BLE: Device 5C:BC:48:E6:B1:48 does not look like a CHIP device.
CHIP:BLE: Device 62:D7:84:68:52:36 does not look like a CHIP device.
CHIP:BLE: Device F3:33:19:1E:96:07 does not look like a CHIP device.
CHIP:BLE: Device 7B:B8:07:F9:B2:DE does not look like a CHIP device.
CHIP:BLE: Device 08:66:98:91:66:E3 does not look like a CHIP device.
CHIP:BLE: Device 1A:F9:BD:D6:E1:4D does not look like a CHIP device.
CHIP:BLE: Device 23:28:86:6C:FF:CD does not look like a CHIP device.
CHIP:BLE: Device 5B:AB:AA:17:D1:6D does not look like a CHIP device.
CHIP:BLE: Device 4F:DA:BD:10:3A:10 does not look like a CHIP device.
CHIP:BLE: Device 26:60:01:44:98:5F does not look like a CHIP device.
[... 155 lines cut ...]
chip-device-ctrl > zcl OnOff Toggle 12344321 1 0
CHIP:ZCL: Successfully encoded 11 bytes
CHIP:IN: Connection from 'UNDEFINED' expired
CHIP:CTL: OnConnectionExpired: Failed to process received message: err 4051
CHIP:IN: New pairing for device 12344321, key 0!!
CHIP:IN: Secure message was encrypted: Msg ID 0
CHIP:IN: Sending msg from 18446743528248704968 to 12344321
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status 0
CHIP:CTL: SendMessage returned 0
chip-device-ctrl > zcl OnOff Toggle 12344321 1 0
CHIP:ZCL: Successfully encoded 11 bytes
CHIP:IN: Secure message was encrypted: Msg ID 1
CHIP:IN: Sending msg from 18446743528248704968 to 12344321
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status 0
CHIP:CTL: SendMessage returned 0
chip-device-ctrl > zcl OnOff Toggle 12344321 1 0
CHIP:ZCL: Successfully encoded 11 bytes
CHIP:IN: Secure message was encrypted: Msg ID 2
CHIP:IN: Sending msg from 18446743528248704968 to 12344321
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status 0
CHIP:CTL: SendMessage returned 0
chip-device-ctrl >

franck-apple, 2021-03-09:
CSG validated that this is needed in the TE1 branch

andy31415, 2021-03-10:
I replaced the label  of cherrypick to just TE1 - I need cherrypick to only apply to PRs (so that I have a commit to cherrypick)

Damian-Nordic, 2021-04-21:
This has been fixed, but the issue hasn't been closed. Closing it now.

closed 2021-04-21 by Damian-Nordic