project-chip/connectedhomeip #5310: raspi accessory could not send BT advertisements

cjandhyala, 2021-03-11:
Seems like commit ID "fec9913efb15c2e212e9e918b8429e59968f7ac2" introduced regression that raspi accessory can't do BT advertisements.. Code prior to this commit is able to send the BT advertisements

$ sudo ./out/host/chip-lighting-app --wifi
SetupPINCode: [12345678]
SetupQRCode:  [CH:I34DV*700 0C9SS0]
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 12344321

CHIP:DL: wpa_supplicant: _IsWiFiStationProvisioned: interface not connected
CHIP:BLE: Waiting for PBKDF param request
CHIP:DIS: CHIP minimal mDNS configured as 'Commisioning device'.
CHIP:DL: MDNS failed to join multicast group on wlan0 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
CHIP:DIS: CHIP minimal mDNS started advertising.
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Broadcasting mDns reply
CHIP:DIS: Broadcasting mDns reply
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Broadcasting mDns reply
CHIP:DIS: Broadcasting mDns reply
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Broadcasting mDns reply
CHIP:DIS: Broadcasting mDns reply
CHIP:SVR: Server Listening...
CHIP:DL: FAIL: BLE name is NULL in ConfigureBluezAdv
CHIP:DL: Disabling CHIPoBLE service due to error: Error 4003 (0x00000FA3)
CHIP:DL: wpa_supplicant: connected to wpa_supplicant proxy
CHIP:DL: wpa_supplicant: WiFi interface: /fi/w1/wpa_supplicant1/Interfaces/0
CHIP:DL: wpa_supplicant: connected to wpa_supplicant interface proxy

vivien-apple, 2021-03-11:
Should be fixed by #5244

vivien-apple, 2021-03-11:
Err. I mean #5312. Please reopen if it does not.

closed 2021-03-11 by vivien-apple