project-chip/connectedhomeip #5328: Generated setup code that is 21 bytes long formatted incorrectly

cjandhyala, 2021-03-11:
Per the spec, 21 bytes setup code should have 1st digit b.w. 4 and 7 if VID and PID are present, Ref attached spec
The current generated setup code seem to have 1st digit as 0 even if the VID and PID are present. 

Also we need to make sure chip-tool is able to parse the new setup code. 
Ex: $ sudo out/host/chip-tool payload parse-setup-payload 039506169709050652794
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:SPL: Parsing decimalRepresentation: 039506169709050652794
CHIP:SPL: RequiresCustomFlow: 1
CHIP:SPL: VendorID: 9050
CHIP:SPL: Version: 0
CHIP:SPL: ProductID: 65279
CHIP:SPL: Discriminator: 0
CHIP:SPL: SetUpPINCode: 12345678
CHIP:SPL: RendezvousInformation: 0

Table 34. Encoding Method with Vendor and Product ID’s included (VID_PID_Present == 1)
[image]

jrhees-cae, 2021-03-11:
Confirmed: the current `ManualSetupPayloadGenerator`:
a) Is combining the fields in a binary fashion and then converting to a decimal. This is incorrect. Instead, each digit range in the table [Digit1][Digit2..6][Digit7..10] should be converted to decimal separately and concatenated.
b) Is not including the version# into the payload at all.

jrhees-cae, 2021-03-11:
This will cause TC#2 sub-items 2, 3, 3b to fail in Test Event 1

closed 2021-03-19 by andy31415