project-chip/connectedhomeip #5331: chip-tool parse setup code should be able to parse concatenated QRcode

cjandhyala, 2021-03-11:
chip-tool parse setup code should be able to parse concatenated QRcode

per section 5.1.5 of the spec, 

[image]
[image]

liamgonyea, 2022-05-25:
This issue needs to be re-opened as there still is support needed for parsing Concatenated QR codes as of TE9 testing timeframe

Rajashreekalmane, 2023-07-18:
Still facing the same issue

cjandhyala, 2023-07-20:
@Rajashreekalmane What is the issue you are facing ? BTW concatenated QR code is not supported in matter 1.2 spec.

closed 2025-12-09 by bzbarsky-apple