project-chip/connectedhomeip #5349: Create a stable (synchronous only) persistent storage interface.

andy31415, 2021-03-12:
#### Problem
Existing persistent storage implementation provides both synchronous and asynchronous APIs.

However actual implementations either implement one side or the other (either sync or async) and usage is not as clear.

 #### Proposed Solution
 
Standardize on a single persistent storage implementation. Synchronous operation seems the easiest to reason about and directly maps to KVS, so that approach can be used.

andy31415, 2021-03-12:
Generally KeyValueStoreManager should be usable across the board EXCEPT likely Android and iOS do not provide KVS.

@rgoliver - is this accurate? Could we have android/iOS/other implementations? Trying to figure out how sane the idea of 'use KVS everywhere' is vs needing one level of indirection to switch between kvs and non-kvs platforms.

rgoliver, 2021-03-16:
I don't know of any technical limitation which would stop KVS being implemented on Android or iOS.

closed 2022-10-09 by stale[bot]