project-chip/connectedhomeip #5382: Add support for M5Stack Core 2 in the CHIP all-clusters-app

sagar-apple, 2021-03-15:
#### Problem
The all-clusters-app for the ESP32 does not seem to fully support the M5Stack Core 2. 

Flashing the app on the M5Stack Core 2 results in a black screen. It appears all other functionality is working though(probably because of the underlying ESP32). 

 #### Proposed Solution
I suspect the display is black because of the display driver we use for the M5Stack. The driver is inverting the display at boot for M5Stack Core (1) support and we probably need to undo that to support the new hardware revision.

dhrishi, 2021-03-17:
@sagar-apple With reference to the comment here, https://github.com/project-chip/connectedhomeip/blob/master/examples/common/screen-framework/include/Display.h#L34-L35 

Can you please try setting `INVERT_DISPLAY` to `INVERT_OFF` and see if that helps?

sagar-apple, 2021-03-18:
Unfortunately that was not it. The display appears completely off. It's possible our display driver isn't compatible with this version of the m5stack.

dhrishi, 2021-03-22:
As per https://m5stack-store.myshopify.com/products/m5stack-core2-esp32-iot-development-kit, the display (LCD screen) used in M5Stack Core 2 is: ILI9342C. The support of this is not present in the ESP32 TFT library which we use in CHIP. (Submodule: examples/common/m5stack-tft/repo => https://github.com/jeremyjh/ESP32_TFT_library.git)

MoeweX, 2021-11-29:
So just to clarify, one has to buy the original basic core, e.g., [M5Stack](https://shop.m5stack.com/products/m5stack-basic-core-esp32-iot-devkit-comm-edition?_pos=1&_sid=149440a4d&_ss=r) rather than the Core 2 to follow the [Apple Enable Matter Instructions](https://developer.apple.com/services-account/download?path=/iOS/iOS_Logs/EnableMatter_Instructions.pdf)?

screensforme, 2022-08-11:
Even I am facing same issue.
M5Stack grey version is EOL and not available for purchase..

Is there any workaround for m5stack core 2 black screen issue ?

closed 2023-02-19 by stale[bot]

schadom, 2023-03-15:
I'm facing the same issue. Have you made any progress on this @sagar-apple , @dhrishi , @MoeweX? Thanks