project-chip/connectedhomeip #5421: Does it have to be paired to be use chip-tool?

yuzibo, 2021-03-17:
#### Problem
On my two PI 4, PI A has launched with cmd:
```bash
ubuntu@pi-2:~/git/connectedhomeip/examples/lighting-app/linux$ sudo ./out/debug/chip-lighting-app --ble 0
SetupPINCode: [12345678]
SetupQRCode:  [CH:I34DV*700 0C9SS0]
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 12344321

CHIP:IN: New pairing for device 112233, key 0!!
CHIP:SVR: Received a new connection.
CHIP:SVR: Rendezvous and secure pairing skipped
CHIP:DIS: Found admin paring for admin 0, node BC5C01
CHIP:DIS: CHIP minimal mDNS configured as 'Operational device'.
CHIP:DIS: CHIP minimal mDNS started advertising.
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Broadcasting mDns reply
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Broadcasting mDns reply
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Broadcasting mDns reply
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Broadcasting mDns reply
CHIP:SVR: Server Listening...
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:DL: TRACE: Bus acquired for name CHIP-3840
CHIP:DL: CREATE service object at /chipoble/0d8f/service
CHIP:DL: Create characteristic object at /chipoble/0d8f/service/c1
CHIP:DL: Create characteristic object at /chipoble/0d8f/service/c2
CHIP:DL: CHIP BTP C1 /chipoble/0d8f/service
CHIP:DL: CHIP BTP C2 /chipoble/0d8f/service
CHIP:DL: CHIP_ENABLE_ADDITIONAL_DATA_ADVERTISING is TRUE
CHIP:DL: Create characteristic object at /chipoble/0d8f/service/c3
CHIP:DL: rotatingDeviceId: 00007C5F6E176CD40F68685D100A1CF8A98B
CHIP:DL: CHIP BTP C3 /chipoble/0d8f/service
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: BluezPeripheralRegisterAppDone done
CHIP:DL: HandlePlatformSpecificBLEEvent 16391
CHIP:DL: Create adv object at /chipoble/0d8f/advertising
CHIP:DL: SET service data to {'0xFEAF': <[byte 0x00, 0x00, 0x0f, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: HandlePlatformSpecificBLEEvent 16392
CHIP:DL: CHIPoBLE advertising config complete
CHIP:DL: RegisterAdvertisement complete
CHIP:DL: HandlePlatformSpecificBLEEvent 16393

```
On PI B using chip-tool, but I tested it many time, I found it must be pairing with

```c
./chip-tool pairing bypass 192.168.1.100 11097
``` 
And test cmd:
```c
./chip-tool onoff on 1
```
Is it right? If it is true, This is clearly not CHIP's intention from my view. I think it is OK for Pi B to getPI A's ip via BLE or other methods other than pairing ip in a local network.

362dyx, 2021-03-18:
yes！You have to execute the pairing command first, and then send the control command, but when I execute the pairing command，have the following log on RTT viewer：
Unencrypted message from UDP:FDB4:EE6:53CC:39C0:0:FF:FE00:4C00:11097 is dropped since no corresponding handler is set in TransportMgr.
Don't you have any similar problems？

yuzibo, 2021-03-18:
Maybe you have missing compiling options --args='chip_bypass_rendzvous=true' in server end.
You can refer to example/lighting-app/linux/README.md.

woody-apple, 2022-01-25:
Is this resolved? Please reopen if not.

closed 2022-01-25 by woody-apple