project-chip/connectedhomeip #5431: CHIPClusters representation for strings does not match command handler respresentation

bzbarsky-apple, 2021-03-17:
#### Problem
Looking at the "lock door" command, CHIPClusters.cpp does:
```
    ReturnErrorOnFailure(writer.PutString(TLV::ContextTag(argSeqNumber++), pin));
```
which won't even compile; `pin` is a `ByteSpan` but there is no `PutString` that takes a `ByteSpan` as far as I can tell.... but if this did the obvious thing, the bytes in the TLV would just be the string data.

But the various command handlers, for ZCL_LOCK_DOOR_COMMAND_ID, do:
```
                    TLVError = dataTlv.GetDataPtr(PIN);
...
            emberAfDoorLockClusterLockDoorCallback(const_cast<uint8_t *>(PIN));
```
which will succeed whether the TLV has a string or octet string (probably buggy), but clearly expects the actual data to be in ZCL string encoding (Pascal string with length at the front), which is just not what CHIPClusters would write.
 

 #### Proposed Solution
Seems like the two ends should match?  And the writer should compile?

@erjiaqing @vivien-apple

erjiaqing, 2021-03-18:
Per cluster XML (and comment about type at the line above)

The pin attribute is CHAR_STRING type, so we use char* instead of ByteSpan.

Previous PRs seems to use ByteSpan for all strings.

I'm courious about the difference between OCTET_STRING and CHAR_STRING.

bzbarsky-apple, 2021-03-18:
In the spec there are two types: "octstr" (defined in the "fundamental types" table) and "string" (defined in the "derived types" table).  The former represents a bag of bytes.  The latter represents a UTF-8 string.  The cluster definition should say which one is in use, but they are NOT the same thing at all.  In particular, they have different TLV encodings, so to the extent that we validate TLV (and we should), one will not decode as the other.  Yes, TLVReader right now treats them as more or less interchangeable....

In any case, we should decide what our representation of "string" will be.  I would vote for `Span<char>`.  And yes, it would not be null-terminated, in either the ZCL encoding or TLV; that's fine.

bzbarsky-apple, 2021-03-18:
That said, for the specific case of door lock, pin codes are "octstr" in the spec, afaict.  I don't know why they would be CHAR_STRING in our XML.

bzbarsky-apple, 2021-03-22:
This is a crititical security issue on the reader because it does not validate that it has all the data the string length claims and hence can cause the cluster code to read random memory.

The only saving grace here is that code is not enabled, but we need to resolve this before enabling it.

erjiaqing, 2021-03-22:
Yes, for this issue, we should update the XML, and the root cause is the different string representation between ZCL & CHIP. I would suggest modify the command handlers and the attributes handlers so it can accept ByteSpan as an argument. #5542

bzbarsky-apple, 2021-12-03:
Fixed a while ago (https://github.com/project-chip/connectedhomeip/pull/10821 etc).

closed 2021-12-03 by bzbarsky-apple