project-chip/connectedhomeip #5455: Debian 11 Python version problems

janusheide, 2021-03-18:
#### Problem

Following the build guide (https://github.com/project-chip/connectedhomeip/blob/master/docs/BUILDING.md) on Debian 11, things does not work out of the box.

I had some problems, and ended going back to try to run the pigweed getting started, that works as long as I set python to point to python3 (python2 is no longer available in Debian 11 and neither is the python command).

Comming back to chip, doing the same thing I get the following error:

```
  WELCOME TO...

  ▄███▒  ░▓█  ░▓█ ░▓█▓ ▒█████▄
 ██▒ ▀█▒  ▒█   ▒█  ░█▒  ▒█░  █░
 █▓░      ▒██████  ░█▒  ▒█▄▄▄█░
 ▓█   █▒  ▒█   ▒█  ░█░  ▒█▀
 ░▓███▀  ░▓███░▓█▒ ░█░  ▒█

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done
  Setting up Python environment.....[|]============================================================
Unexpected Python version: Python 3.9.1+
============================================================
failed
Error during bootstrap--see messages above.
```

Do you plan to pull any changes from the pigweed build scripts, which might make this work on Debian 11?

 #### Proposed Solution
1. Pull changes to the install scripts from pigweed.
2. Relax the requirements to the installed Python version.

janusheide, 2021-03-18:
I dug around a bit more and found that the dbus-python package was the main problem as it is not available under Python3.9, nor is it maintained for what looks like a year, and the authors discourage its use: https://pypi.org/project/dbus-python/#description

So I would suggest to remove/replace this dependency, it does not look like it is very extensively used.

https://github.com/project-chip/connectedhomeip/blob/f58003ca2671e50d24b0db0ea595ec851f7cdebf/scripts/requirements.txt#L33

mattyang2012, 2021-05-05:
@janusheide : I'm running into the same issue.  Could you elaborate about how you work around this issue? 
```
$source ./scripts/activate.sh

  WELCOME TO...

  ▄███▒  ░▓█  ░▓█ ░▓█▓ ▒█████▄
 ██▒ ▀█▒  ▒█   ▒█  ░█▒  ▒█░  █░
 █▓░      ▒██████  ░█▒  ▒█▄▄▄█░
 ▓█   █▒  ▒█   ▒█  ░█░  ▒█▀
 ░▓███▀  ░▓███░▓█▒ ░█░  ▒█

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done
  Setting up Python environment.....[|]============================================================
Unexpected Python version: Python 3.9.2
============================================================
failed
Error during bootstrap--see messages above.```

janusheide, 2021-05-05:
I just ran that stuff inside a virtual machine (ubuntu), this problem could be fixed if the upstream was followed, so I assume it will go away eventually.

If you really want you can remove the python version check, remove the dependency, but then you are left with a few places where that module is used which you also need to remove...

janusheide, 2021-05-05:
I just pulled the newest master and tried again and did not bump into the problem.

So maybe you can try to pull the latest master?

If you get python version error, you can try to add the following to your .bashrc

alias python="/usr/bin/python3"

mattyang2012, 2021-05-06:
@janusheide : I pulled the newest master and `activate.sh` works. Thanks a lot!

janusheide, 2021-05-06:
Sweet, Ill close this one then :)

closed 2021-05-06 by janusheide