project-chip/connectedhomeip #5493: Public SendMessage APIs that don't allow passing message header info make no sense

bzbarsky-apple, 2021-03-19:
#### Problem
We have some APIs that look broken to me:
```
Device::SendMessage(System::PacketBufferHandle buffer)
SecureSessionMgr::SendMessage(SecureSessionHandle session, System::PacketBufferHandle && msgBuf)
```
These are broken because they do not say what type of message (protocol id, protocol opcode) to send.

 #### Proposed Solution
1. Stop having message-sending APIs that don't take type-of-message info, either as a `PayloadHeader` or explicitly (like exchange context does).  Fix tests that use such APIs to use an API that does take this information.
2. Add a constructor that takes protocol id and message type to `PayloadHeader`.
3. Ensure that all on-stack construction of `PayloadHeader` uses that constructor, so the only place where `PayloadHeader` is default-constructed is in the various structs that store it for resend purposes.  Thought even there we might be able to give those structs better "is constructed" semantics that would avoid the need for uninitialized payload headers.

@pan-apple @yufengwangca

yufengwangca, 2021-03-19:
Those APIs exist before Exchange Layer is introduced, we should stop using those APIs, but we first need to port all legacy examples to talk to messaging layer instead of transport layer directly.

bzbarsky-apple, 2021-03-19:
We can maybe do these in parallel, to the extent that we have consumers of these APIs at all...

yufengwangca, 2021-04-11:
This issue is fixed in #5627

closed 2021-04-11 by yufengwangca

bzbarsky-apple, 2021-04-13:
Well, one of the three items that needed to be fixed is fixed...

I filed https://github.com/project-chip/connectedhomeip/issues/5984 on what's left.