project-chip/connectedhomeip #5518: [Linux] Unable to connect chip-lighting-app(server) and chip-tool(client) over ble

rohit10000, 2021-03-20:
#### Problem
Facing connection error when trying to connect connectedhomeip/examples/lighting-app/linux (server) with connectedhomeip/examples/chip-tool (client) over ble.

 * **expected behaviour:** server and client should connect over ble successfully
 * **actual behaviour:** Fail to connect device, connection refused.
 * **system configuration:** Ubuntu 20.04
 * **screenshots, images, logs, etc.**

**Logs for server:**
~/connectedhomeip/examples/lighting-app/linux/out/debug$ ./chip-lighting-app --ble-device 0
SetupPINCode: [12345678]
SetupQRCode:  [CH:I34DV*700 0C9SS0]
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 12344321

CHIP:DL: wpa_supplicant: _IsWiFiStationProvisioned: interface not connected
CHIP:BLE: Waiting for PBKDF param request
CHIP:DL: MDNS failed to join multicast group on enp1s0 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
CHIP:DIS: CHIP minimal mDNS started advertising.
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:SVR: Server Listening...
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:DL: TRACE: Bus acquired for name CHIP-3840
CHIP:DL: CREATE service object at /chipoble/63e4/service
CHIP:DL: Create characteristic object at /chipoble/63e4/service/c1
CHIP:DL: Create characteristic object at /chipoble/63e4/service/c2
CHIP:DL: CHIP BTP C1 /chipoble/63e4/service
CHIP:DL: CHIP BTP C2 /chipoble/63e4/service
CHIP:DL: CHIP_ENABLE_ADDITIONAL_DATA_ADVERTISING is TRUE
CHIP:DL: Create characteristic object at /chipoble/63e4/service/c3
CHIP:DL: rotatingDeviceId: 00007C5F6E176CD40F68685D100A1CF8A98B
CHIP:DL: CHIP BTP C3 /chipoble/63e4/service
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: BluezPeripheralRegisterAppDone done
CHIP:DL: HandlePlatformSpecificBLEEvent 16391
CHIP:DL: Create adv object at /chipoble/63e4/advertising
CHIP:DL: SET service data to {'0xFEAF': <[byte 0x00, 0x00, 0x0f, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: HandlePlatformSpecificBLEEvent 16392
CHIP:DL: CHIPoBLE advertising config complete
CHIP:DL: RegisterAdvertisement complete
CHIP:DL: HandlePlatformSpecificBLEEvent 16393

**Logs for Client:**
./chip-tool pairing ble CHIP-3840 3840 12345678 3840
CHIP:ZCL: Using ZAP configuration...
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233
CHIP:ZCL: Using ZAP configuration...
CHIP:DL: CHIP task running
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:DL: TRACE: Bus acquired for name C-d473
CHIP:DL: PlatformBlueZInit init success
CHIP:BLE: BLE scanning through known devices.
CHIP:BLE: Device F9:72:11:1E:1D:D8 does not look like a CHIP device.
CHIP:BLE: Device FC:58:FA:A3:64:0D does not look like a CHIP device.
CHIP:BLE: Device 70:74:FA:71:A5:D1 does not look like a CHIP device.
CHIP:BLE: Device 5A:6A:7D:3E:44:93 does not look like a CHIP device.
CHIP:BLE: Device 7B:E5:2F:AB:9C:C1 does not look like a CHIP device.
CHIP:BLE: Device 77:7B:C6:A6:50:67 does not look like a CHIP device.
CHIP:BLE: Device FC:58:FA:07:C5:08 does not look like a CHIP device.
CHIP:BLE: Device 55:A5:2E:9B:28:01 does not look like a CHIP device.
CHIP:BLE: Device 7D:33:5B:E6:9D:5C does not look like a CHIP device.
CHIP:BLE: Device 6F:DB:AF:66:20:42 does not look like a CHIP device.
CHIP:BLE: Device 51:2D:32:4E:0C:D8 does not look like a CHIP device.
CHIP:BLE: Device 5A:0D:9C:21:2A:96 does not look like a CHIP device.
CHIP:BLE: Device 20:A6:0C:AB:00:A9 does not look like a CHIP device.
CHIP:BLE: Device 57:8B:B4:2E:39:DE does not look like a CHIP device.
CHIP:BLE: Device 66:32:A6:C2:45:E3 does not look like a CHIP device.
CHIP:BLE: Device 77:17:A0:1D:F5:C2 does not look like a CHIP device.
CHIP:BLE: Device 4A:0D:5E:51:31:78 does not look like a CHIP device.
CHIP:BLE: Device 70:BB:E9:60:6E:5B does not look like a CHIP device.
CHIP:BLE: Device 47:4D:D7:52:64:4E does not look like a CHIP device.
CHIP:BLE: Device 45:C1:6D:39:29:F1 does not look like a CHIP device.
CHIP:BLE: Device 5C:F9:3C:1C:3F:E2 does not look like a CHIP device.
CHIP:BLE: Device 53:29:96:B6:50:51 does not look like a CHIP device.
CHIP:BLE: Device 47:43:6E:18:80:46 does not look like a CHIP device.
CHIP:BLE: Device 7B:CE:98:69:19:D0 does not look like a CHIP device.
CHIP:BLE: New device scanned: D0:7E:35:D0:A7:18
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Failed to stop discovery GDBus.Error:org.bluez.Error.Failed: No discovery started
CHIP:BLE: Scan complete notification without an active scan.
CHIP:BLE: Device 6E:DF:0B:B5:86:7B does not look like a CHIP device.
CHIP:BLE: Device 53:A8:1F:F9:25:08 does not look like a CHIP device.
CHIP:BLE: Device 48:83:B4:18:50:DC does not look like a CHIP device.
CHIP:BLE: Device 6C:20:F3:DE:73:D6 does not look like a CHIP device.
CHIP:BLE: Device 51:28:DA:D5:E7:EF does not look like a CHIP device.
CHIP:BLE: Device 52:A5:2D:E4:53:35 does not look like a CHIP device.
CHIP:BLE: Device 53:5D:18:CD:39:04 does not look like a CHIP device.
CHIP:BLE: Device 68:2A:71:13:45:CF does not look like a CHIP device.
CHIP:BLE: Device 47:AE:54:E7:4E:92 does not look like a CHIP device.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: Failed to start discovery: Operation was cancelled
CHIP:BLE: Scan complete notification without an active scan.
**CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.Failed: Connection refused**
CHIP:TOO: No response from device
CHIP:CTL: Shutting down the commissioner
CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
CHIP:CTL: Shutting down the controller
CHIP:DL: System Layer shutdown
CHIP:DL: Inet Layer shutdown
CHIP:DL: BLE layer shutdown

Any suggestion about how to solve the issue will be helpful.

 #### Proposed Solution
<suggested fix, suggested enhancement>

Damian-Nordic, 2021-03-26:
Are you running the client on RPi? Anyway, have you tried to run `systemctl restart bluetooth` to restart the Bluez daemon? Not sure why that happens, but sometimes Bluez enters a weird state in which in can connect no device.

rohit10000, 2021-03-26:
I was running the client and server both on Ubuntu 20.04LTS differently.

woody-apple, 2022-01-25:
Given this is an old issue, is this still present? If so, please re-open, thanks!

closed 2022-01-25 by woody-apple