project-chip/connectedhomeip #5541: Re-enable CodeQL

woody-apple, 2021-03-22:
#### Problem
CodeQL has been failing almost every build due to "out of space".

#### Proposed Solution
Fix and re-enable

woody-apple, 2021-03-22:
Looks like:
```
              with:
                  fetch-depth: 0
```
Is checking out all branches, which... is huge.

woody-apple, 2021-03-22:
Looks like this is needed for bloat check, is there a way to just pull master (needed branches?) instead of all the branches?

Damian-Nordic, 2021-03-29:
I wonder if `fetch-depth: 0` is still needed. It was added in https://github.com/project-chip/connectedhomeip/commit/67a1b5919, but at that time the bloat check was executed in the examples' workflows and now it's performed in a separate job.

cecille, 2022-02-01:
@kpschoedel (I think you worked on bloat check) - any idea about this one?

kpschoedel, 2022-02-01:
Bloat check no longer needs it. (It now records sizes of individual builds from CI and matches them up with parent builds later.)

closed 2022-12-20 by stale[bot]