project-chip/connectedhomeip #5556: rework system layer event loop

mspang, 2021-03-22:
#### Problem

The code in src/{system,inet} and src/platform/{Linux,Darwin} is based on use of the select(2) system call.

Unfortunately, select(2) is incompatible with event loops used in many modern programs, e.g. those based on abstractions such as those provided by libevent / libev / libuv or that directly use low level event based facilities such as Linux's epoll(7).

The main issue is that these event loops are based on persistent event subscriptions, while select(2) has no persistent state and instead requires calling each entity that's interested in polling for every main loop iteration. This is a major impedance mismatch that complicates integrating CHIP into programs.

Another, lesser problem is the use of fd_sets which imposes a relatively low limit on file descriptor value.

 #### Proposed Solution

- Generalize the system layer to support modern event loops
- Adopt a suitable event loop abstraction for in-repo examples and tests. e.g. libevent
- Strawman Alternative: Use libuv, which also provides a straightforward path to native Windows support.

Ideal outcome - CHIP's IO thread can be removed, and tests/examples are all single threaded

mspang, 2021-03-22:
@cecille @kpschoedel @turon

mspang, 2021-03-22:
@tcarmelveilleux

turon, 2021-03-24:
I agree that moving Linux to a more event based model that aligns with RTOS platform abstractions would be ideal.
The issue is how to do this in a way that is easy to integrate into apps that are based on the various Posix mainloop models (such as select and poll variants).

cecille, 2021-03-24:
@mlepage-google - what are your thoughts on libuv for this usecase?

mspang, 2021-03-29:
@turon Agree we should strive to make it easy to use whatever main loop implementation the application uses natively (whether that be be libevent, libev, libuv, glib, etc). Or even select() if the application really does use that.

I believe it will be much easier to map from an event style interface onto a select() backend than the reverse. The problem with going from select() to epoll() is that we are not notified of new sources as soon as we become interesting.. but rather the next time around the loop. So it's effectively impossible to integrate CHIP into these systems without introducing a new dedicated thread for it to run on.

closed 2022-10-18 by stale[bot]