project-chip/connectedhomeip #5561: Compiling bridge-app demo on Raspberry Pi 4b error

ChengqiangShe, 2021-03-23:
#### 
Compiling bridge-app demo on Raspberry Pi 4b error
System environment: Ubuntu 20.10 
commit：e5fb65262951717adc964f880bdcdcfab471e780

The error is as follows:
Checking the environment:

20210323 11:18:06 INF Environment passes all checks!

Environment looks good, you are ready to go!

Done. Made 74 targets from 78 files in 449ms
ninja: Entering directory `out/debug'
[268/378] c++ obj/third_party/connectedhomeip/src/messaging/libMessagingLayer.ExchangeMgr.cpp.o
FAILED: obj/third_party/connectedhomeip/src/messaging/libMessagingLayer.ExchangeMgr.cpp.o 
g++ -MMD -MF obj/third_party/connectedhomeip/src/messaging/libMessagingLayer.ExchangeMgr.cpp.o.d -Wconversion -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Wextra -Wshadow -Wunreachable-code -Werror -Wno-deprecated-declarations -Wno-unknown-warning-option -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wno-cast-function-type -fdiagnostics-color -fno-strict-aliasing -D_REENTRANT -isystem../../third_party/connectedhomeip/third_party/ot-br-posix/repo/src -isystem../../third_party/connectedhomeip/third_party/ot-br-posix/repo/include -I/usr/include/dbus-1.0 -I/usr/lib/aarch64-linux-gnu/dbus-1.0/include -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -DLWIP_IPV4=1 -DLWIP_IPV6=1 -DLWIP_API=1 -DLWIP_ETHERNET=1 -DLWIP_SLIP=0 -DLWIP_6LOWPAN=1 -DLWIP_PPP=1 -I../../third_party/connectedhomeip/src/include -I../../third_party/connectedhomeip/src -Igen/include -I../../third_party/connectedhomeip/src/lib -I../../third_party/connectedhomeip/config/standalone -I../../third_party/connectedhomeip/third_party/nlassert/repo/include -I../../third_party/connectedhomeip/third_party/nlio/repo/include -I../../third_party/connectedhomeip/src/lwip/standalone -I../../third_party/connectedhomeip/third_party/lwip/repo/lwip/src/include -I../../third_party/connectedhomeip/src/lwip/include -I../../third_party/connectedhomeip/src/setup_payload -Igen -Igen -I../../third_party/connectedhomeip/third_party/inipp/repo/inipp -c ../../third_party/connectedhomeip/src/messaging/ExchangeMgr.cpp -o obj/third_party/connectedhomeip/src/messaging/libMessagingLayer.ExchangeMgr.cpp.o
../../third_party/connectedhomeip/src/messaging/ExchangeMgr.cpp: In member function ‘CHIP_ERROR chip::Messaging::ExchangeManager::Init(chip::NodeId, chip::TransportMgrBase*, chip::SecureSessionMgr*)’:
../../third_party/connectedhomeip/src/messaging/ExchangeMgr.cpp:83:51: error: ‘void* memset(void*, int, size_t)’ clearing an object of non-trivial type ‘struct chip::Messaging::ExchangeManager::UnsolicitedMessageHandler’; use assignment or value-initialization instead [-Werror=class-memaccess]
   83 |     memset(UMHandlerPool, 0, sizeof(UMHandlerPool));
      |                                                   ^
In file included from ../../third_party/connectedhomeip/src/messaging/ExchangeMgr.cpp:39:
../../third_party/connectedhomeip/src/messaging/ExchangeMgr.h:232:12: note: ‘struct chip::Messaging::ExchangeManager::UnsolicitedMessageHandler’ declared here
  232 |     struct UnsolicitedMessageHandler
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
[273/378] c++ obj/third_party/connectedhomeip/src/messaging/libMessagingLayer.Channel.cpp.o
ninja: build stopped: subcommand failed.

ChengqiangShe, 2021-03-23:
In the same compilation environment, fall back to commit fc5d1b7bdd497c5f3f7fdb3c534c433429c3099e
Compiled successfully

mspang, 2021-03-24:
Already fixed.

closed 2021-03-24 by mspang