project-chip/connectedhomeip #5589: Cannot pairing chip device by chip-tool or chip-device-ctrl because of "GDBus.Error:org.bluez.Error.Failed: Input/output error"

ElthamLiu, 2021-03-24:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior: chip-tool or chip-device-ctrl and connect and pair chip device
 * actual behavior: chip-tool and chip-device-ctrl connot throw error "GDBus.Error:org.bluez.Error.Failed: Input/output error"
 * steps to reproduce
 * system configuration: Raspberry Pi 4b
 * screenshots, images, logs, etc.
 >
ubuntu@ubuntu:~/connectedhomeip$ sudo chip-device-ctrl 
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 18446743528248704968

CHIP:DL: CHIP task running
Chip Device Controller Shell

chip-device-ctrl > set-pairing-thread-credential 11 1234 11223344556677889900aabbccddeeff
Thread credential set
chip-device-ctrl > connect -ble 3840 12345678 1234
Device is assigned with nodeid = 1234
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:DL: TRACE: Bus acquired for name C-0808
CHIP:DL: PlatformBlueZInit init success
CHIP:BLE: BLE scanning through known devices.
CHIP:BLE: Device 00:0B:57:A8:FC:81 does not look like a CHIP device.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: Device 00:17:DD:08:F7:6D does not look like a CHIP device.
CHIP:BLE: Device 00:17:DD:08:2A:F0 does not look like a CHIP device.
CHIP:BLE: Device 68:0A:E2:38:6A:E5 does not look like a CHIP device.
CHIP:BLE: Device 00:17:DD:08:38:B0 does not look like a CHIP device.
CHIP:BLE: Device 00:17:DD:08:1C:E7 does not look like a CHIP device.
CHIP:BLE: Device 68:0A:E2:38:70:45 does not look like a CHIP device.
CHIP:BLE: Device 3C:6A:9D:04:41:0D does not look like a CHIP device.
CHIP:BLE: Device 51:2B:E3:E6:84:3A does not look like a CHIP device.
CHIP:BLE: Device 00:17:DD:08:30:03 does not look like a CHIP device.
CHIP:BLE: Device 77:13:1A:76:DF:B8 does not look like a CHIP device.
CHIP:BLE: Device 00:17:DD:08:87:55 does not look like a CHIP device.
CHIP:BLE: Device F5:93:8D:2D:59:00 does not look like a CHIP device.
CHIP:BLE: Device 00:17:DD:08:1C:EE does not look like a CHIP device.
CHIP:BLE: Device 78:75:4E:D2:B3:9B does not look like a CHIP device.
CHIP:BLE: Device 00:17:DD:08:F7:6D does not look like a CHIP device.
CHIP:BLE: Device 60:34:D3:FF:3A:D3 does not look like a CHIP device.
CHIP:BLE: Device 5B:55:CA:3E:F4:33 does not look like a CHIP device.
CHIP:BLE: Device 72:B4:8D:FC:F9:92 does not look like a CHIP device.
CHIP:BLE: Device 61:90:A0:AC:46:6D does not look like a CHIP device.
CHIP:BLE: Device E4:6C:99:18:E2:F3 does not look like a CHIP device.
CHIP:BLE: Device 00:17:DD:08:0B:05 does not look like a CHIP device.
CHIP:BLE: Device 3C:6A:9D:04:41:0D does not look like a CHIP device.
CHIP:BLE: Device 7C:0F:27:B8:83:0A does not look like a CHIP device.
CHIP:BLE: New device scanned: 00:0B:57:A8:FC:81
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.Failed: Input/output error

It seems like the bluez module on my Raspberry Pi has some problem, does anyone met this issue? How to fix it?
Thanks

 #### Proposed Solution
<suggested fix, suggested enhancement>

ElthamLiu, 2021-03-24:
After I run "systemctl restart bluetooth*" command, bluetooth module work fine, but I still has some problem when try to pair chip device (lighting-app run on EFR32MG12), chip device show "InvalidState" during commission process:

00> <detail> [DL] Thread joiner timer running

00> <detail> [DL] Request Thread joiner start

00> <info  > [DL] Joiner Discerner: 3840

00> <info  > [DL] Joiner PSKd: 012345678

00> <info  > [DL] Joiner start: No Error

00> <info  > [DL] _OnPlatformEvent default:  event->Type = 32778

00> <detail> [DL] OpenThread State Changed (Flags: 0x0800800b)

00> <detail> [DL]    Network Name: OpenThread

00> <detail> [DL]    PAN Id: 0x64F4

00> <detail> [DL]    Extended PAN Id: 0xDEAD00BEEF00CAFE

00> <detail> [DL]    Channel: 11

00> <detail> [DL]    Mesh Prefix: FDDE:AD00:BEEF::/64

00> <detail> [DL]    Thread Unicast Addresses:

00> <detail> [DL]         FE80::30A6:2A3:55C0:7189/64 valid preferred

00> <detail> [DL] LwIP Thread interface addresses updated

00> <detail> [DL]    FE80::30A6:2A3:55C0:7189 IPv6 link-local address, preferred)

00> <info  > [DL] Connect Event for handle : 1

00> <info  > [DL] Char Write Req, char : 26

00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 9)

00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived

00> <info  > [BLE] local and remote recv window sizes = 3

00> <info  > [BLE] selected BTP version 3

00> <info  > [BLE] using BTP fragment sizes rx 128 / tx 20.

00> <info  > [DL] CHIPoBLE subscribe received

00> <info  > [DL] _OnPlatformEvent kCHIPoBLESubscribe

00> <info  > [DL] _OnPlatformEvent default:  event->Type = 32777

00> <info  > [DL] Tx Confirmation received

00> <info  > [DL]  stop soft timer

00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm

00> <info  > [DL] Char Write Req, char : 26

00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 67)

00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived

00> <info  > [BLE] Received pairing message for lu

00> <detail> [BLE] Received PBKDF param request

[... 296 lines cut ...]
**00> <info  > [DL] Joiner start: OpenThread Error 9000013 (0x0089544D): InvalidState**

00> <info  > [DL] Char Write Req, char : 26

00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)

00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived

00> <info  > [DL] Tx Confirmation received

00> <info  > [DL]  stop soft timer

00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm

00> <info  > [DL] Char Write Req, char : 26

00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)

00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived

00> <info  > [DL] Tx Confirmation received

00> <info  > [DL]  stop soft timer

00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm

JasonHu1, 2021-07-19:
I have the same problem，Do you fix it?

woody-apple, 2022-01-25:
Given this is an old issue, is this still present? If so, please re-open, thanks!

closed 2022-01-25 by woody-apple