project-chip/connectedhomeip #5602: Android project is not compiling

virupaksha, 2021-03-24:
On my macbook pro I am facing error when I run "TARGET_CPU=arm64 ./scripts/examples/android_app.sh".

Below are the logs which I see at the end. Can someone help me to solve this issue?

//*********** Logs **************//

ninja -C out/android_arm64 src/setup_payload/java src/controller/java default
ninja: Entering directory `out/android_arm64'
ninja: no work to do.
rsync -a out/android_arm64/lib/CHIPController.jar out/android_arm64/lib/SetupPayloadParser.jar src/android/CHIPTool/app/libs
rsync -a out/android_arm64/lib/jni/arm64-v8a src/android/CHIPTool/app/src/main/jniLibs
git submodule update --init --recursive third_party/ot-commissioner/repo
TARGET_CPU=arm64
./third_party/ot-commissioner/build-android-libs.sh
./third_party/ot-commissioner/build-android-libs.sh: line 3: realpath: command not found
CMake Error: The source directory "/Users/virupakshfutane/Desktop/CHIP/repo" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
//***********************//

Damian-Nordic, 2021-03-26:
@virupaksha as the error indicates, you don't have `realpath` on your system. Could you modify `third_party/ot-commissioner/build-android-libs.sh` by replacing `realpath -s` with `readlink -f` and see if that helps?

virupaksha, 2021-04-02:
Hi @Damian-Nordic  thank you for reply. I solved my problem by "brew install coreutils".

closed 2021-05-05 by virupaksha