project-chip/connectedhomeip #5604: Python controller support to read  Application Basic Cluster info

cjandhyala, 2021-03-25:
Python controller support to read  Application Basic Cluster info, Per 6.1.1.7 Test Procedure of Device Management and Section of 8.2. of the chip protocol spec. 

[image]

Damian-Nordic, 2021-04-09:
@franck-apple are you sure it's needed for TE2? There are two clusters of similar names: Basic Cluster and Application Basic Cluster and thought that the former would be needed for TE2 based on the "CSG Test Event Execution Plan" document. Can this be clarified?

woody-apple, 2021-04-13:
@Damian-Nordic TE2 is focusing on this item in the spec: "11.2. Basic Information Cluster"

woody-apple, 2021-04-15:
@Damian-Nordic  - just for clarity, @cjandhyala's reference is to an older part of the spec, this is indeed required for TE2, and the spec is the 11.2. Basic Information Cluster.

Damian-Nordic, 2021-04-15:
@woody-apple Ok, thanks for clarification!

cjandhyala, 2021-04-18:
@Damian-Nordic , Here is the reference for the test case https://github.com/project-chip/chip-test-plans/blob/master/src/devicemanagement.adoc#6117-test-procedure , we should be able read all these attributes per test plan

Damian-Nordic, 2021-04-19:
@cjandhyala Added two PRs which together should allow you to read all Basic Cluster attributes.

Examples of reading mandatory attributes:
```
chip-device-ctrl > zclread Basic VendorName 1 1 1
...
CHIP:ZCL:   value: TEST_VENDOR

chip-device-ctrl > zclread Basic VendorID 1 1 1
...
CHIP:ZCL:   value: 0x235a

chip-device-ctrl > zclread Basic ProductName 1 1 1
...
CHIP:ZCL:   value: TEST_PRODUCT

chip-device-ctrl > zclread Basic ProductID 1 1 1
...
CHIP:ZCL:   value: 0x4e4b

chip-device-ctrl > zclread Basic UserLabel 1 1 1
...
CHIP:ZCL:   value: 

chip-device-ctrl > zclread Basic Location 1 1 1
...
CHIP:ZCL:   value:

chip-device-ctrl > zclread Basic HardwareVersion 1 1 1
...
CHIP:ZCL:   value: 0x0001

chip-device-ctrl > zclread Basic HardwareVersionString 1 1 1
...
CHIP:ZCL:   value: TEST_VERSION

chip-device-ctrl > zclread Basic SoftwareVersion 1 1 1
...
CHIP:ZCL:   value: 0x00000001

chip-device-ctrl > zclread Basic SoftwareVersionString 1 1 1
...
CHIP:ZCL:   value: prerelease
```

The remaining attributes are optional and if a given attribute is not supported by the DUT, the command will return the following error:
```
chip-device-ctrl > zclread Basic ManufacturingDate 1 1 1
CHIP:ZCL: ReadAttributesResponse:
CHIP:ZCL:   ClusterId: 0x0028
CHIP:ZCL:   attributeId: 0x000b
CHIP:ZCL:   status: EMBER_ZCL_STATUS_UNSUPPORTED_ATTRIBUTE (0x86)
Failed to establish secure session to device: 134
CHIP:ZCL: Data model processing success!
An exception occurred during reading ZCL attribute:
```

franck-apple, 2021-04-22:
Some of the optional attributes are missing. Python controller is currently able to parse all the mandatory attributes, but is not able to validate optional attributes if an accessory vendor implements it.

Damian-Nordic, 2021-04-23:
@franck-apple can someone provide any logs? They worked last time I tried.

cjandhyala, 2021-04-23:
@Damian-Nordic per our slack conversation the other day, right now python controller can parse the mandatory attributes, which is working fine. We need python controller to be able to read the remaining optional attributes as well in case any vendor implements those.

Damian-Nordic, 2021-06-16:
As discussed on slack, Python CHIP Controller can already read all Basic cluster attributes. If there's a need to enable the optional attributes on a specific example, another issue should be created.

PR #6135 (2021-04-20): [zcl] Enable optional Basic Cluster attributes for controller apps
 #### Problem
Controller apps should provide support for reading Basic Cluster attributes marked as optional even though they may not be available on a specific device.

 #### Summary of Changes
* Enable optional Basic Cluster attributes in ZAP files for controller apps
* Re-generate ZCL code

**Note: you can just review the first commit, as the second one contains just auto-generated code**

 Fixes #5604
closed 2021-06-16 by Damian-Nordic