project-chip/connectedhomeip #5619: Full network commissioning cluster support

erjiaqing, 2021-03-25:
#### Problem

Currently we only implemented 3 most urgent commands for network commissioning cluster. We need a full network commissioning cluster support which also makes use of full IM features.

 #### Proposed Solution
<suggested fix, suggested enhancement>

kkasperczyk-no, 2021-04-21:
@erjiaqing it seems that on accessory side we are missing handling some NetworkCommissioning commands like ScanNetworks, RemoveNetwork, DisableNetwork and GetLastNetworkCommissioningResult that are planned to be used on TE2. Are you working actively on any of that? I planned to try implement it (at least some of those starting from Scan), but I noticed that this issue exists and I thought that we should sync to avoid duplicating our work.

erjiaqing, 2021-04-21:
> @erjiaqing it seems that on accessory side we are missing handling some NetworkCommissioning commands like ScanNetworks, RemoveNetwork, DisableNetwork and GetLastNetworkCommissioningResult that are [...]

I'm current enhancing the Add(WiFi|Thread)Network and EnableNetwork commands, and a pr is expected in very soon.
I'm not current working on Scan command.

erjiaqing, 2021-07-15:
As network commissioning contains a set of complex commands, the following issues need to be addressed before:
1. The CommandHandler should support async command handling. (https://github.com/project-chip/connectedhomeip/pull/8197 is trying to solve this)
2. The command payload should support complex types. (#2182, may require ZAP update)

erjiaqing, 2021-11-17:
As of Nov 21' the implementing of full network commissioning can continue. Based on current obersivations of the code.

- The plan is to continue using DeviceNetworkProvisioningDelegate, with some changes
  - It will be renamed to `DeviceNetworkCommissioningDelegate` to match the name of network commissioning cluster
  - I'm going to add a timeout param as well as a callback (object) param to handle network commissioning related events, for each of the network operations (Scan, Connect(Join), Disconnect)
  - The network commissioning won't use DeviceNetworkProvisioningDelegateImpl, instead, DeviceNetworkProvisioningDelegate will be used as an interface
  - Basic compatibility will be provided by "DefaultDeviceNetworkCommissioningDelegate" before platforms implemented the new interface.
  - The full network commissiong PR will be presented with a Linux version of DeviceNetworkProvisioningDelegate.

erjiaqing, 2021-12-21:
Fixed, follow up: https://github.com/project-chip/connectedhomeip/issues/13170 (Platform driver)

closed 2021-12-21 by erjiaqing