project-chip/connectedhomeip #5622: chip-tool: uninitialized value comparison

andy31415, 2021-03-25:
#### Problem
 Ran chip-tool commisioning under valgrind, got an error about jump depending on uninitialized value:
 
 ```
 CHIP:IN: New pairing for device 12344321, key 0!!                                                                                                                                                                                    
==1496058== Thread 7:                                                                                                                                                                                                                
==1496058== Conditional jump or move depends on uninitialised value(s)                                                                                                                                                               
==1496058==    at 0x1A48C0: chip::Inet::IPAddress::operator!=(chip::Inet::IPAddress const&) const (IPAddress.cpp:55)                                                                                                                 
==1496058==    by 0x1D8248: chip::SecureSessionMgr::NewPairing(chip::Optional<chip::Transport::PeerAddress> const&, unsigned long, chip::PASESession*, chip::SecureSessionMgr::PairingDirection, unsigned short, chip::Transport::Bas
e*) (SecureSessionMgr.cpp:262)                                                                                                                                                                                                       
==1496058==    by 0x1D53ED: chip::RendezvousSession::OnSessionEstablished() (RendezvousSession.cpp:188)                                                                                                                              
==1496058==    by 0x1D3F8C: chip::PASESession::HandleMsg2_and_SendMsg3(chip::PacketHeader const&, chip::System::PacketBufferHandle const&) (PASESession.cpp:689)                                                                     
==1496058==    by 0x1D4424: chip::PASESession::HandlePeerMessage(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) (PASESession.cpp:810)                                             
==1496058==    by 0x1D5BB3: chip::RendezvousSession::HandlePairingMessage(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) (RendezvousSession.cpp:386)                              
==1496058==    by 0x1D59F9: chip::RendezvousSession::OnRendezvousMessageReceived(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle) (RendezvousSession.cpp:358)                       
==1496058==    by 0x1DA022: chip::Transport::BLE::OnBleEndPointReceive(chip::Ble::BLEEndPoint*, chip::System::PacketBufferHandle) (BLE.cpp:192)                                                                                      
==1496058==    by 0x1FC9AB: chip::Ble::BLEEndPoint::Receive(chip::System::PacketBufferHandle) (BLEEndPoint.cpp:1433)
==1496058==    by 0x1FE13E: chip::Ble::BleLayer::HandleIndicationReceived(void*, chip::Ble::ChipBleUUID const*, chip::Ble::ChipBleUUID const*, chip::System::PacketBufferHandle) (BleLayer.cpp:517)
==1496058==    by 0x1B2D64: chip::DeviceLayer::Internal::BLEManagerImpl::HandlePlatformSpecificBLEEvent(chip::DeviceLayer::ChipDeviceEvent const*) (BLEManagerImpl.cpp:305)
==1496058==    by 0x1B2B56: chip::DeviceLayer::Internal::BLEManagerImpl::_OnPlatformEvent(chip::DeviceLayer::ChipDeviceEvent const*) (BLEManagerImpl.cpp:267)
==1496058== 
CHIP:CTL: Remote device completed SPAKE2+ handshake
```

kpschoedel, 2022-01-31:
probably obsolete

woody-apple, 2022-02-07:
This does appear to be resolved

closed 2022-02-07 by woody-apple