project-chip/connectedhomeip #5637: ZAP: Remove profile id from endpoint

lazarkov, 2021-03-26:
#### Problem
Currently you can add/edit a Profile ID on a endpoint through zap tool. 
Profile ID is not supported on CHIP. It creates only confusion. 

 #### Proposed Solution
Remove the profile id reference from the ZAP tool.
Remove the profile id reference from the generated code.

lazarkov, 2021-03-26:
[
[image]
](url)

bzbarsky-apple, 2021-03-26:
This will need some changes to non-generated code too, because some of that examines the profile id.

bzbarsky-apple, 2021-05-22:
> Is profileID meaningless for CHIP, or does it simply get set to a specific value and nobody should touch it?

Meaningless; CHIP has different extension mechanisms.

woody-apple, 2022-01-25:
Given this is an old issue, is this still present? If so, please re-open, thanks!

closed 2022-01-25 by woody-apple

bzbarsky-apple, 2022-01-26:
Moved to https://github.com/project-chip/zap/issues/394