project-chip/connectedhomeip #5678: [Crash] System Layer crash: Wrong handler! in chip::System::Layer::HandleEvent

sagar-apple, 2021-03-30:
#### Problem
Sending repeated random commands to Level Control Cluster causes the M5Stack to crash sometimes. 
Stack trace below.  
 
 ```
 V (87329) chip[ZCL]: Measured APS frame size 11
V (87329) chip[ZCL]: Successfully encoded 11 bytes
V (87339) chip[IN]: Secure message was encrypted: Msg ID 36
I (87339) chip[IN]: Sending msg from 3 to 9777222158381001781
I (87349) chip[IN]: Sending secure msg on generic transport
I (87349) chip[IN]: Secure msg send status 0
I (87359) chip[ZCL]: T   0:TX (0x3f40cb26) Ucast 0x00x3f400c5f
I (87369) chip[ZCL]: TX buffer: [
I (87369) chip[ZCL]: 08 1F 0B 00 00 
I (87369) chip[ZCL]: ]
V (87379) chip[ZCL]: Data model processing success!
E (87379) chip[DL]: Long dispatch time: 151 ms
I (87389) chip[ZCL]: Event: move from 94
I (87389) chip[ZCL]:  to 0 
I (87389) chip[ZCL]: (diff -1)
I (87399) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0008', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (87409) app-devicecallbacks: Unhandled cluster ID: 8
I (87419) app-devicecallbacks: Current free heap: 114104

I (87419) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0005', EndPoint ID: '0x01', Attribute ID: '0x0003'
I (87429) app-devicecallbacks: Unhandled cluster ID: 5
I (87439) app-devicecallbacks: Current free heap: 114104

I (87449) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0008', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (87459) app-devicecallbacks: Unhandled cluster ID: 8
I (87459) app-devicecallbacks: Current free heap: 114104

V (87469) chip[CSL]: wrong handler! [target 0x0 != this 0x3ffce8c8]
abort() was called at PC 0x40137e8b on core 1
0x40137e8b: chip::System::Layer::HandleEvent(chip::System::Object&, int, unsigned int) at /Users/sagar_dhawan/Work/connectedhomeip/examples/all-clusters-app/esp32/build/chip/../../../../../config/esp32/third_party/connectedhomeip/src/system/SystemLayer.cpp:888

ELF file SHA256: 4ad6548a1a9b4af0

Backtrace: 0x4008e7c8:0x3ffdcc10 0x4008ebe5:0x3ffdcc30 0x40137e8b:0x3ffdcc50 0x40136495:0x3ffdcc70 0x401364e6:0x3ffdcc90 0x40136707:0x3ffdccb0 0x40136731:0x3ffdcd10 0x4008f371:0x3ffdcd30
0x4008e7c8: invoke_abort at /Users/sagar_dhawan/Work/misc/esp-idf/components/esp32/panic.c:159

0x4008ebe5: abort at /Users/sagar_dhawan/Work/misc/esp-idf/components/esp32/panic.c:174

0x40137e8b: chip::System::Layer::HandleEvent(chip::System::Object&, int, unsigned int) at /Users/sagar_dhawan/Work/connectedhomeip/examples/all-clusters-app/esp32/build/chip/../../../../../config/esp32/third_party/connectedhomeip/src/system/SystemLayer.cpp:888

0x40136495: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToSystemLayer(chip::DeviceLayer::ChipDeviceEvent const*) at /Users/sagar_dhawan/Work/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.cpp:248

0x401364e6: chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) at /Users/sagar_dhawan/Work/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl.cpp:208

0x40136707: chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at /Users/sagar_dhawan/Work/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:234
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() at /Users/sagar_dhawan/Work/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:175

0x40136731: chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /Users/sagar_dhawan/Work/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/include/platform/PlatformManager.h:204
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /Users/sagar_dhawan/Work/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.cpp:198

0x4008f371: vPortTaskWrapper at /Users/sagar_dhawan/Work/misc/esp-idf/components/freertos/port.c:143

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7036
load:0x40078000,len:13752
load:0x40080400,len:3868
entry 0x40080680
I (29) boot: ESP-IDF v4.1-289-g167966380-dirty 2nd stage bootloader
I (29) boot: compile time 10:44:51
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 001e6400
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2c804 (182276) map
I (182) esp_image: segment 1: paddr=0x0003c82c vaddr=0x3ffbdb60 size=0x037ec ( 14316) load
I (188) esp_image: segment 2: paddr=0x00040020 vaddr=0x400d0020 size=0xb6f78 (749432) map
0x400d0020: _stext at ??:?

I (472) esp_image: segment 3: paddr=0x000f6fa0 vaddr=0x3ffc134c size=0x03770 ( 14192) load
I (478) esp_image: segment 4: paddr=0x000fa718 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/sagar_dhawan/Work/misc/esp-idf/components/freertos/xtensa_vectors.S:1778
```

woody-apple, 2021-04-12:
@sweetymhaiske would you be able to take a look at this crash? Thank you!

sweetymhaiske, 2021-04-14:
> @sweetymhaiske would you be able to take a look at this crash? Thank you!

@woody-apple Sure. I am looking into it.

sweetymhaiske, 2021-04-14:
@sagar-apple 
I tried with recent commits on master branch. As I do not have M5Stack board in hand, I tried it on DevkitC board However I choose Device Type as (M5Stack) from menuconfig.
I sent 104 commands on `levelcontrol` cluster using `chip-tool`, however, I haven't faced such issue.

Can you please try with the recent commits on master branch?
In case you still faced the issue, then can you please mention the steps to reproduce it?

sagar-apple, 2021-04-22:
@sweetymhaiske I tried this again with ToT from this morning. 

I'm still seeing this crash when send level changes in quick succession. 

Here's how I reproduced it. 
I made the following modification to CHIPTool on iOS. 
```
--- a/src/darwin/CHIPTool/CHIPTool/View Controllers/OnOffCluster/OnOffViewController.m  
+++ b/src/darwin/CHIPTool/CHIPTool/View Controllers/OnOffCluster/OnOffViewController.m  
@@ -239,12 +239,17 @@ NSString * const kCHIPNumLightOnOffCluster = @"OnOffViewController_NumLights";
     [_deviceSelector forSelectedDevices:^(uint64_t deviceId) {
         CHIPDevice * chipDevice = CHIPGetPairedDeviceWithID(deviceId);
         if (chipDevice != nil) {
-            CHIPOnOff * onOff = [[CHIPOnOff alloc] initWithDevice:chipDevice endpoint:endpoint queue:dispatch_get_main_queue()];
-            [onOff on:^(NSError * error, NSDictionary * values) {
-                NSString * resultString
-                    = (error != nil) ? [NSString stringWithFormat:@"An error occured: 0x%02lx", error.code] : @"On command success";
+            CHIPLevelControl * level = [[CHIPLevelControl alloc] initWithDevice:chipDevice endpoint:endpoint queue:dispatch_get_main_queue()];
+            [level moveToLevelWithOnOff:(int8_t)arc4random_uniform(255) transitionTime:0 completionHandler:^(NSError * _Nullable error, NSDictionary * _Nullable values) {
+                NSString * resultString = (error != nil) ? [NSString stringWithFormat:@"An error occured: 0x%02lx", error.code] : @"On command success";
                 [self updateResult:resultString];
             }];
```
That quick hack allows me to randomly send a level change command to the accessory every time I tap on the Light "On" button in the CHIPTool iOS app's On Off Cluster. I tapped on it in quick succession a few times and the M5stack crashed and rebooted.

sweetymhaiske, 2021-04-29:
@sagar-apple Yes, the issue reproduced.