project-chip/connectedhomeip #5679: Cirque CI break on MobileDeviceTest test case

yufengwangca, 2021-03-30:
#### Problem

Cirque CI breaks on my PR: https://github.com/project-chip/connectedhomeip/pull/5627 within test case MobileDeviceTest. After I revert my change and run Cirque on Tip of master, I saw same error:

Traceback (most recent call last):
  File "/home/yufengw/Workspace/connectedhomeip/src/test_driver/linux-cirque/helper/CHIPTestBase.py", line 62, in run_test
    self.test_routine()
  File "/home/yufengw/Workspace/connectedhomeip/src/test_driver/linux-cirque/test-mobile-device.py", line 65, in test_routine
    self.run_controller_test()
  File "/home/yufengw/Workspace/connectedhomeip/src/test_driver/linux-cirque/test-mobile-device.py", line 93, in run_controller_test
    self.assertTrue('leader' in roles)
  File "/home/yufengw/Workspace/connectedhomeip/src/test_driver/linux-cirque/helper/CHIPTestBase.py", line 210, in assertTrue
    raise AssertionError
AssertionError
2021-03-29 19:17:36,493 [CHIPInteractionModelTest] INFO destroying home: 64a9a4a3-fedc-4eee-9bdf-bd938f4268a6
Cleanup Flask
./scripts/tests/cirque_tests.sh: line 43: kill: (84248) - No such process
./scripts/tests/cirque_tests.sh: line 43: kill: (84253) - No such process
./scripts/tests/cirque_tests.sh: line 43: kill: (84258) - No such process
Test log can be found at /tmp/tmp.qjw51YQkQm/MobileDeviceTest/device_logs
yufengw@yufengw-intel:~/Workspace/connectedhomeip$ 

Suspect this is regression from https://github.com/project-chip/connectedhomeip/pull/5317

 #### Proposed Solution
<suggested fix, suggested enhancement>

erjiaqing, 2021-03-30:
After taking some look at the log, 

```

Segmentation fault (core dumped)
*** Segmentation fault
Register dump:

 RAX: 2d8292e87f8d12bd   RBX: 000000000218a680   RCX: 0000000000000000
 RDX: 0000000000000000   RSI: 0000000000000000   RDI: 000000000218d568
 RBP: 00007f54377fd700   R8 : 00007f5447951f6d   R9 : 000000000000001e
 R10: 00007f5447a58070   R11: 0000000000000000   R12: 000000000218d618
 R13: 0000000000000000   R14: 0000000000000001   R15: 0000000000000000
 RSP: 00007f54377fd6f0

 RIP: 00007f54479b26e4   EFLAGS: 00010202

 CS: 0033   FS: 0000   GS: 0000

 Trap: 0000000d   Error: 00000000   OldMask: 00000000   CR2: 00000000

 FPUCW: 0000037f   FPUSW: 00000000   TAG: 00000000
 RIP: 00000000   RDP: 00000000

 ST(0) 0000 0000000000000000   ST(1) 0000 0000000000000000
 ST(2) 0000 0000000000000000   ST(3) 0000 0000000000000000
 ST(4) 0000 0000000000000000   ST(5) 0000 0000000000000000
 ST(6) 0000 0000000000000000   ST(7) 0000 0000000000000000
 mxcsr: 1fa0
 XMM0:  00000000000000000000000025252525 XMM1:  00000000000000000000000025252525
 XMM2:  00000000000000000000000025252525 XMM3:  00000000000000000000000025252525
 XMM4:  00000000000000000000000025252525 XMM5:  00000000000000000000000025252525
 XMM6:  00000000000000000000000025252525 XMM7:  00000000000000000000000025252525
 XMM8:  00000000000000000000000025252525 XMM9:  00000000000000000000000025252525
 XMM10: 00000000000000000000000025252525 XMM11: 00000000000000000000000025252525
 XMM12: 00000000000000000000000025252525 XMM13: 00000000000000000000000025252525
 XMM14: 00000000000000000000000025252525 XMM15: 00000000000000000000000025252525

Backtrace:
/usr/local/lib/python3.8/dist-packages/chip/_ChipDeviceCtrl.so(_ZN4chip17RendezvousSession11UpdateStateENS0_5StateEi+0x184)[0x7f54479b26e4]
```

According to the code, I suspect RendezvousSession is used after freed. @pan-apple PTAL.

yufengwangca, 2021-03-30:
It looks all Cirque CI go back to normal after reverting #5317

closed 2021-03-30 by yufengwangca