project-chip/connectedhomeip #5685: Unable to scan Nordic Dev Board flashed with chip lighting sample app using chip device controller

SaileeG, 2021-03-30:
Unable to scan Nordic Dev Board flashed with chip lighting sample app using chip device controller

I confirmed that the BLE Advertisements are enabled from:
Board Logs:
[1146081667] <inf> app: NFC Tag emulation is already started
[1146081691] <dbg> chip.LogV: 690335637[DL] SetAdvertisingEnabled(true)
[1146081718] <inf> app: Enabled BLE Advertisement
[1146081749] <dbg> chip.LogV: 690335639[DL] In DriveBLEState
[1146081827] <inf> chip: 690335641[DL] CHIPoBLE advertising started
[1146946019] <inf> app: NFC Tag emulation is already started
[1146946041] <inf> app: BLE Advertisement is already enabled
 
and Nrf-connect app on Mobile:
Advertisement data from nrf connect Apphttps://drive.google.com/file/d/1R8PKNML-WmN2YwkTQBamDe655AvpGIwy/view?usp=sharing
using the nrf connect app I am able to scan and connect to the nordic board running Chip Lighting app.

Ad an offline Discussion with @Song and he mentioned the board does not seem to broadcast its service uuid list which might be the issue. @Damian-Nordic can you help take a look

SaileeG, 2021-03-30:
I am running the Chip device controller on Raspi running ubuntu

erjiaqing, 2021-03-30:
> Strange, I'm using Python CHIP controller on Linux and it can find Nordic devices. Also, I'm pretty sure we tested the Darwin implementation in the past and it was working. Which OS are you using?

Seems the device is filtered

```
    def find_devices(self, uuids):
        devices = [
            BluezDbusDevice(p["object"], self.bluez, self.bus, self.logger)
            for p in get_bluez_objects(
                self.bluez, self.bus, DEVICE_INTERFACE, self.path
            )
        ]
        found = []
        for device in devices:
            for i in device.uuids:
                if i in uuids:
                    found.append(device)
                    break
```

The bluez scan api will use the service uuid list in broadcast data to filter devices.

Damian-Nordic, 2021-03-30:
I just built Python CHIP controller from the current master and was able to find the Nordic lock-app:
```
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:DL: CHIP task running
Chip Device Controller Shell

chip-device-ctrl > ble-scan -t 10
2021-03-30 11:59:15,932 ChipBLEMgr   INFO     use default adapter
2021-03-30 11:59:16,142 ChipBLEMgr   INFO     scanning started
2021-03-30 11:59:16,370 ChipBLEMgr   INFO     Name            = ChipLock
2021-03-30 11:59:16,371 ChipBLEMgr   INFO     ID              = 2538b9f5-6d0a-3663-83a6-1ee88311cdab
2021-03-30 11:59:16,372 ChipBLEMgr   INFO     RSSI            = -58
2021-03-30 11:59:16,374 ChipBLEMgr   INFO     Address         = 7E:E0:82:8E:FE:65
2021-03-30 11:59:16,377 ChipBLEMgr   INFO     Pairing State   = 0
2021-03-30 11:59:16,378 ChipBLEMgr   INFO     Discriminator   = 3840
2021-03-30 11:59:16,378 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-03-30 11:59:16,378 ChipBLEMgr   INFO     Product Id      = 20043
2021-03-30 11:59:16,381 ChipBLEMgr   INFO     Adv UUID        = 0000feaf-0000-1000-8000-00805f9b34fb
2021-03-30 11:59:16,382 ChipBLEMgr   INFO     Adv Data        = 00000f5a234b4e
2021-03-30 11:59:16,382 ChipBLEMgr   INFO
```
I'm assuming you test the master, too, right? That's true that on the Nordic platform the service uuid list is not broadcast, but AFAIK it's not required by the spec, so the controller should required that either.

SaileeG, 2021-03-30:
I am running from Master as well and using the Lighting App

Damian-Nordic, 2021-03-31:
@SaileeG Could you paste logs from the controller after you run `ble-scan` command? @erjiaqing is right that in the mentioned piece of code a Nordic DK will be filtered out, but besides that part a native BLE scanner is called, which should successfully discover the device.

SaileeG, 2021-04-05:
The issue is seen only when we have BLE virtual interfaces running on the Host.

If I stop all virtual interfaces and kill all bluetoothd processes the I am able to scan the Nordic dev boards.
Running chip device controller with HCI0 with no Virtual BLE interfaces and Killall bluetoothd

```
chip-device-ctrl > ble-scan
2021-04-05 06:01:25,715 ChipBLEMgr   INFO     scanning started
2021-04-05 06:01:26,137 ChipBLEMgr   INFO     Name            = ChipLight
2021-04-05 06:01:26,138 ChipBLEMgr   INFO     ID              = f016e23d-0d00-35d5-93e7-588acdbc7e54
2021-04-05 06:01:26,139 ChipBLEMgr   INFO     RSSI            = -57
2021-04-05 06:01:26,141 ChipBLEMgr   INFO     Address         = D4:15:B1:19:B0:2E
2021-04-05 06:01:26,144 ChipBLEMgr   INFO     Pairing State   = 0
2021-04-05 06:01:26,145 ChipBLEMgr   INFO     Discriminator   = 3840
2021-04-05 06:01:26,145 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-04-05 06:01:26,145 ChipBLEMgr   INFO     Product Id      = 65279
2021-04-05 06:01:26,148 ChipBLEMgr   INFO     Adv UUID        = 0000feaf-0000-1000-8000-00805f9b34fb
2021-04-05 06:01:26,149 ChipBLEMgr   INFO     Adv Data        = 00000f5a23fffe
2021-04-05 06:01:26,149 ChipBLEMgr   INFO     
2021-04-05 06:01:26,519 ChipBLEMgr   INFO     Name            = ChipLight
2021-04-05 06:01:26,520 ChipBLEMgr   INFO     ID              = f016e23d-0d00-35d5-93e7-588acdbc7e54
2021-04-05 06:01:26,521 ChipBLEMgr   INFO     RSSI            = -35
2021-04-05 06:01:26,523 ChipBLEMgr   INFO     Address         = D3:E0:81:02:64:3F
2021-04-05 06:01:26,527 ChipBLEMgr   INFO     Pairing State   = 0
2021-04-05 06:01:26,527 ChipBLEMgr   INFO     Discriminator   = 3840
2021-04-05 06:01:26,527 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-04-05 06:01:26,527 ChipBLEMgr   INFO     Product Id      = 65279
2021-04-05 06:01:26,530 ChipBLEMgr   INFO     Adv UUID        = 0000feaf-0000-1000-8000-00805f9b34fb
2021-04-05 06:01:26,532 ChipBLEMgr   INFO     Adv Data        = 00000f5a23fffe
2021-04-05 06:01:26,532 ChipBLEMgr   INFO     

2021-04-05 06:01:35,883 ChipBLEMgr   INFO     scanning stopped
 
chip-device-ctrl > ble-adapter-print
2021-04-05 06:01:40,227 ChipBLEMgr   INFO     adapter 0 = B8:27:EB:C4:E2:93
```
Without BLE virtual interfaces We are unable to scan Linux Lighting app running on the same host with error seen on Linux Lighting App side:  'CHIP:DL: FAIL: device is not connected'

closed 2021-04-07 by mspang