project-chip/connectedhomeip #5746: Add the compiled CHIPTool apk file

JumaLuata, 2021-04-01:
It is a bit complicated for non-Android developers to set up the CHIP Tool compilation environment.  Compilation issues I encountered: [#5481](https://github.com/project-chip/connectedhomeip/issues/5481)
Is it possible to put the compiled APK file into repository? This will be much more friendly.

lukgni, 2021-04-05:
Some workaround for your issue may be adding artifact creation in existing Android GitHub workflow,
similar to what we've done in our Arm Mbed fork.
(https://github.com/ARMmbed/connectedhomeip/pull/130)
 
If you want to build apk file manually and only when it's needed, you may try to reuse chip-android container which suits that purpose perfectly. 
Here you can find detailed instruction: 
https://github.com/ARMmbed/connectedhomeip/wiki/The-easiest-way-to-build-Android-CHIPTool-app

JumaLuata, 2021-04-06:
I got the APK files from [Android  GitHub workflow Artifacts of Arm Mbed fork](https://github.com/ARMmbed/connectedhomeip/actions/runs/721462717).
Thank you @lukgni .

woody-apple, 2022-01-25:
Given this is an older issue, does this till reproduce on current builds? Please re-open if so!

closed 2022-01-25 by woody-apple

ZBXallen, 2023-03-30:
> I got the APK files from [Android GitHub workflow Artifacts of Arm Mbed fork](https://github.com/ARMmbed/connectedhomeip/actions/runs/721462717). Thank you @lukgni .

Hi,JumaLuata. 
Can you send me this apk file,because I can't get it from your link. Thanks.