project-chip/connectedhomeip #5771: MessageCounterSyncMgr::RetransTableEntry should perhaps store more data in the packet buffer

bzbarsky-apple, 2021-04-02:
#### Problem
`MessageCounterSyncMgr::RetransTableEntry ` has these members:
```
        ExchangeContext * exchangeContext; /**< The ExchangeContext for the stored CHIP message.
                                                Non-null if and only if this entry is in use. */
        System::PacketBufferHandle msgBuf; /**< A handle to the PacketBuffer object holding the CHIP message. */
        SendFlags sendFlags;               /**< Flags set by the application for the CHIP message being sent. */
        Protocols::Id protocolId;          /**< The protocol identifier of the CHIP message to be sent. */
        uint8_t msgType;                   /**< The message type of the CHIP message to be sent. */
```
We could store the protocolId and msgType inside the packetbuffer by endoding a PacketHeader in there, so we don't need this out-of-band storage.  It might even be possible to store the sendFlags in there as well.  Need to check.

 #### Proposed Solution
Move state out of the entry into the packetbuffer.

@yufengwangca @pan-apple

bzbarsky-apple, 2021-04-08:
To be clear, the proposal is to temporarily encode the data into the buffer's header area while the buffer is in the retrans table, and pull it out from there when we go to actually retransmit.

yufengwangca, 2021-04-13:
This change can only reduce 32 bytes of RAM, given the codesize and rodata increases, and the small size of the bss win here, close this issue.

closed 2021-04-13 by yufengwangca