project-chip/connectedhomeip #5782: [nrfconnect] nrfconnect light example failed with NoBufs

wgtdkp, 2021-04-03:
#### Problem
The provisioning nrfconnect light by following https://github.com/project-chip/connectedhomeip/blob/master/examples/lighting-app/nrfconnect/README.md is failing with openthread `kNoBufs` error.

Commit ID: 4f0ade5fcd50f99f53188ec401e18c2a22f47e14

Logs:
```txt
I: 20625 [BLE]Received pairing message for 0x0000000000000013
D: 20630 [BLE]Received spake2p msg3
E: 20634 [BLE]Missing node id in rendezvous parameters. Node ID is required until opcerts are implemented
D: 20643 [IN]New pairing for device 0x0000000000000000, key 0!!
I: 20650 [SVR]Received a new connection.
I: 20654 [SVR]Device completed SPAKE2+ handshake
D: 23084 [DL]Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 23091 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 3)
D: 23174 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 25694 [DL]Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 27270 [DL]Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
D: 27277 [DL]Sending indication for CHIPoBLE TX characteristic (ConnId 0, len 3)
D: 27359 [DL]Indication for CHIPoBLE TX characteristic done (ConnId 0x00, result 0x00)
D: 27368 [DL]Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
I: 27375 [BLE]Received rendezvous message for 0x0000000000000013
I: 27381 [BLE]Received rendezvous message from  0x000000000001b669
I: 27387 [NP]Received kThreadAssociationRequest
E: not word-aligned: 0x000f801e:36
E: 27395 [NP]Failed in HandleNetworkProvisioningMessage. error OpenThread Error 9000003 (0x00895443): NoBufs

E: 27405 [SVR]OnRendezvousStatusUpdate: OpenThread Error 9000003 (0x00895443): NoBufs
D: 27413 [DL]SetAdvertisingEnabled(false)
I: 27417 [SVR]Persisting admin ID 0, next available 1
E: not word-aligned: 0x000f8046:16
I: 27425 [BLE]Releasing end point's BLE connection back to application.
I: 27431 [SVR]OnRendezvousError: OpenThread Error 9000003 (0x00895443): NoBufs
I: 27438 [SVR]Persisting admin ID 0, next available 1
E: not word-aligned: 0x000f8056:16
I: 27447 [SVR]OnRendezvousError: OpenThread Error 9000003 (0x00895443): NoBufs
D: 27454 [DL]In DriveBLEState
D: 27457 [DL]In DriveBLEState
D: 29834 [DL]Write request received for CHIPoBLE RX characteristic (ConnId 0x00)
I: 29841 [BLE]Releasing end point's BLE connection back to application.
E: 29847 [BLE]HandleChipConnectionReceived failed, err = 6018
E: 29853 [BLE]failed handle new chip BLE connection, status = 6018
D: 42343 [DL]ConnId: 0x00, New CCCD value: 0x0000
E: 42348 [BLE]no endpoint for unsub recvd
I: 43423 [DL]BLE GAP connection terminated (reason 0x13)
I: 43429 [DL]Current number of connections: 0/1
D: 43433 [DL]In DriveBLEState
```

wgtdkp, 2021-04-06:
@Damian-Nordic @LuDuda Do you have any thoughts?

Damian-Nordic, 2021-04-06:
@wgtdkp Our colleague had similar issue and it was caused by programming a new firmware version without fully erasing the previously installed one. Could you try `west flash --erase` or `nrfjprog -e`?

wgtdkp, 2021-04-06:
Thanks @Damian-Nordic `nrfjprog -e`  works for me!
Should we document this in https://github.com/project-chip/connectedhomeip/blob/master/examples/lighting-app/nrfconnect/README.md for other developers?

Damian-Nordic, 2021-04-06:
Yes, I guess we can recommend using `west flash --erase` instead of bare `west flash`.

mspang, 2021-04-06:
I don't think this is closed. Was the flash corrupted, and if so how did it get that way? Would be good too understand what happened here...

mspang, 2021-04-06:
This started at

a3d3d6ee ("[nrfconnect] Updated nRF Connect SDK revision (#5613)")

mspang, 2021-04-06:
@rgoliver Any idea about the alignment error?

mspang, 2021-04-06:
Nevermind @rgoliver , this uses Zephyr settings subsystem not pw_kvs

LuDuda, 2021-04-25:
@mspang @wgtdkp just FYI we have identified the problem (different alignment configuration caused by changing the flash driver between NCS revision). We're investigating how to prevent such issue in the future.

Damian-Nordic, 2021-04-30:
To summarize this issue:
1. It was caused by updating NCS from 1.5 release to master, which introduced a new flash driver. The old flash driver supported the write block size equal to 1B while the new flash driver writes data in 4B blocks.
2. The code for initialization of the non-volatile storage in Zephyr was not able to correctly load flash entries written by a driver that has different parameters than the current one.
3. I prepared a fix for that to Zephyr: https://github.com/zephyrproject-rtos/zephyr/pull/34592. It will be included in NCS 1.6, so that upgrades from NCS 1.5 to NCS 1.6 will not result in the error you encountered.

@mspang Thanks for making us look into this issue. Can it be closed now?

PR #5808 (2021-04-06): [nrfconnect] Recommend using west flash --erase
 #### Problem
Using bare `west flash`, without erasing the entire flash
area, may lead to weird behaviour when switching between
different nRF Connect SDK/Zephyr/OpenThread revisions.
It's safer and faster to use `west flash --erase`.

 #### Summary of Changes
Recommend using `west flash --erase` in nRF Connect examples documentation.

 Fixes #5782
closed 2021-07-21 by Damian-Nordic