project-chip/connectedhomeip #5801: chip controller fails to pair with raspi Ubuntu server accessory with BT error code 6007

cjandhyala, 2021-04-05:
Chip controller running in iOS fails to pair with raspi accessory running on Ubuntu server (20.10 64 bit) with BT error code 6007. Ref controller log named 'with-ubuntu-server-log.txt' , device log named  'raspi-ubuntu-server-log.txt'

Whereas the same controller pairs fine with raspi accessory running on Ubuntu Desktop (64 bit).  Ref log named 'with-ubuntu-desktop-log.txt' device log named 'raspi-ubuntu-desktop-log.txt'

[raspi-ubuntu-server-log.txt](https://github.com/project-chip/connectedhomeip/files/6260975/raspi-ubuntu-server-log.txt)
[raspi-ubuntu-desktop-log.txt](https://github.com/project-chip/connectedhomeip/files/6260976/raspi-ubuntu-desktop-log.txt)
[with-ubuntu-server-log.txt](https://github.com/project-chip/connectedhomeip/files/6260977/with-ubuntu-server-log.txt)
[with-ubuntu-desktop-log.txt](https://github.com/project-chip/connectedhomeip/files/6260978/with-ubuntu-desktop-log.txt)

yunhanw-google, 2021-04-06:
Hi, @Damian-Nordic , do you have bandwidth on this issue? thanks

Damian-Nordic, 2021-04-06:
@yunhanw-google I'll take a look today.

Damian-Nordic, 2021-04-08:
Seems like there's some problem with WPA configuration and the commissioned device is unable to join a network:
```
CHIP:NP: LinuxNetworkProvisioningDelegate: SSID: Temp
CHIP:DL: wpa_supplicant: added network: SSID: Temp: /fi/w1/wpa_supplicant1/Interfaces/0/Networks/2
CHIP:DL: wpa_supplicant: connected to network: SSID: Temp
CHIP:DL: wpa_supplicant: failed to save config: GDBus.Error:fi.w1.wpa_supplicant1.UnknownError: Not allowed to update configuration (update_config=0)
CHIP:BLE: Indication confirmation, 0xffff90046710
RTNETLINK answers: Operation not permitted
mkdir: cannot create directory '/run/systemd/resolve/netif': Permission denied
CHIP:DL: Failed to run dhclient, system() returns 256
CHIP:SVR: Device was assigned network credentials
CHIP:DL: MDNS failed to join multicast group on wlan0 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
```

Can you try to fix the permission issue on your system and try again?

cjandhyala, 2021-04-08:
In both the cases app started with 'sudo' previlages 'sudo ./out/host/chip-lighting-app --wifi' , 

I see the same 'CHIP:DL: wpa_supplicant: failed to save config: GDBus.Error:fi.w1..wpa_supplicant1.UnknownError: Not allowed to update configuration (update_config=0)' in both the Ubuntu desktop and Ubuntu server logs attached. There could be a different reason for the BT error on ubuntu server image.

yunhanw-google, 2021-04-08:
From @yufengwangca , regarding WiFi supplicant permission in Ubuntu server, you need do as below
By default, wpa_supplicant is not allowed to update (overwrite) configuration, if you want chip app to be able to store the configuration changes permanently, we need to make the following changes.
Edit the dbus-fi.w1.wpa_supplicant1.service file to use configuration file instead.
sudo nano /etc/systemd/system/dbus-fi.w1.wpa_supplicant1.service
Change the wpa_supplicant start parameters to:
ExecStart=/sbin/wpa_supplicant -u -s -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf
Add the wpa-supplicant configuration file
sudo nano /etc/wpa_supplicant/wpa_supplicant.conf
And add the following content to the file:
ctrl_interface=DIR=/run/wpa_supplicant
update_config=1
Finally, reboot your RPi

Damian-Nordic, 2021-04-09:
@cjandhyala Maybe there is another reason for the BT error, but first we need to solve this one as on Ubuntu Server there's no log message: `Got IP address on interface: wlan0 IP: 192.168.6.116` which means that the rendezvous process is not moved forward and maybe just at some point it times out.

cjandhyala, 2021-04-13:
@yufengwangca , the changes didn't it make it any better.. attached log.
[raspi_ubunutu_server_log_afterupdatingWPASupplicant.txt](https://github.com/project-chip/connectedhomeip/files/6306106/raspi_ubunutu_server_log_afterupdatingWPASupplicant.txt)

closed 2022-01-25 by woody-apple