project-chip/connectedhomeip #5818: ESP32 idf.py build broken for all-clusters app

bzbarsky-apple, 2021-04-06:
#### Problem
If I follow the steps in examples/all-clusters-app/esp32/README.md to try to build the all-clusters-app on ESP32, I get:
```
In file included from examples/common/screen-framework/include/Screen.h:28,
                 from examples/common/screen-framework/include/ListScreen.h:28,
                 from examples/common/screen-framework/ListScreen.cpp:26:
examples/common/screen-framework/include/Display.h:51:10: fatal error: tft.h: No such file or directory
 #include "tft.h"
          ^~~~~~~
compilation terminated.
```

 #### Proposed Solution
Following the directions should work.

For comparison, using the "old" way to build (via `source idf.sh` and then `idf make` works just fine.  The officially documented way should also work.

@dhrishi @sweetymhaiske

sweetymhaiske, 2021-04-07:
@bzbarsky-apple I rechecked it. It worked fine on my side. Can you please clean and rebuild via CMake?

dhrishi, 2021-04-07:
@sweetymhaiske The above compilation seems to be for M5Stack and yes, it is failing. Can you please check?

sweetymhaiske, 2021-04-07:
@dhrishi @bzbarsky-apple  I compiled for ESP32, for which it worked fine. Yes for M5Stack it was failing. I have added the fixes for the same.

closed 2021-04-09 by andy31415

bzbarsky-apple, 2021-04-10:
Thank you for the quick fix!  Sorry, I should have specified I was compiling for m5stack...  I did try cleaning to make sure that was not the problem, of course.  :)

PR #5834 (2021-04-09): ESP32: Fix for CMake build failure for M5Stack and rebuild.
 #### Problem
1. The CMake build failed for M5Stack.
2. The CMake rebuild failure:-
    After build if we make any changes in CHIP code, it won't get reflect or compiled. We need to clean the complete build to look for the changes.

 #### Summary of Changes
1. Added the needed changes in `/examples/common/screen-framework/CMakeLists.txt`
2. Added the fix under `config/esp32/components/chip/CMakeLists.txt`

 Fixes #5818