project-chip/connectedhomeip #5852: chip-device-ctrl is case sensitive, it shouldn't be

bluebin14, 2021-04-08:
#### Problem
chip-device-ctrl fails to parse commands unless arbitrary case sensitivity is followed.

chip-device-ctrl > zcl onoff on 99999 1 0
An exception occurred during process ZCL command:
UnknownCluster: onoff

chip-device-ctrl > zcl OnOff On 99999 1 0
CHIP: [ZCL] Successfully encoded 11 bytes
CHIP: [IN] Secure message was encrypted: Msg ID 3
CHIP: [IN] Sending msg from 0x000000000001b669to 0x000000000001869f
CHIP: [IN] Sending secure msg on generic transport
CHIP: [IN] Secure msg send status 0
CHIP: [CTL] SendMessage returned 0

 #### Proposed Solution
Fix argument parsing so it is not case sensitive.

yunhanw-google, 2021-06-02:
Would you like to take this work? @bluebin14 ?

bluebin14, 2022-07-25:
Closing due to lack of activity.

closed 2022-07-25 by bluebin14