project-chip/connectedhomeip #5875: ReliableMessageContext is too big

bzbarsky-apple, 2021-04-08:
ReliableMessageContext is 56 bytes on 64-bit, 36 bytes on 32-bit.  We have 8 of these, one per ExchangeContext.

@andy31415 did a a clang AST dump for this class on 64-bit, which looks like this:
```
         0 |   class chip::Messaging::ReliableMessageContext mReliableMessageContext
         0 |     class chip::BitFlags<enum chip::Messaging::ReliableMessageContext::Flags, unsigned short> mFlags
         0 |       unsigned short mValue
         8 |     class chip::Messaging::ReliableMessageMgr * mManager
        16 |     class chip::Messaging::ExchangeContext * mExchange
        24 |     class chip::Messaging::ReliableMessageDelegate * mDelegate
        32 |     struct chip::Messaging::ReliableMessageProtocolConfig mConfig
        32 |       uint32_t mInitialRetransTimeoutTick
        36 |       uint32_t mActiveRetransTimeoutTick
        40 |       uint16_t mAckPiggybackTimeoutTick
        42 |       uint8_t mMaxRetrans
        44 |     uint16_t mNextAckTimeTick
        48 |     uint32_t mPendingPeerAckId
```

(that adds up to 52, but this class has 8-byte alignment on 64-bit, so we get 4 bytes of padding).

On 32-bit, it would look like this:

```
         0 |   class chip::Messaging::ReliableMessageContext mReliableMessageContext
         0 |     class chip::BitFlags<enum chip::Messaging::ReliableMessageContext::Flags, unsigned short> mFlags
         0 |       unsigned short mValue
         4 |     class chip::Messaging::ReliableMessageMgr * mManager
         8 |     class chip::Messaging::ExchangeContext * mExchange
        12 |     class chip::Messaging::ReliableMessageDelegate * mDelegate
        16 |     struct chip::Messaging::ReliableMessageProtocolConfig mConfig
        16 |       uint32_t mInitialRetransTimeoutTick
        20 |       uint32_t mActiveRetransTimeoutTick
        24 |       uint16_t mAckPiggybackTimeoutTick
        26 |       uint8_t mMaxRetrans
        28 |     uint16_t mNextAckTimeTick
        32 |     uint32_t mPendingPeerAckId
```

Going to focus on the 32-bit case here.  Some things to think about:

* [ ] We end up with 2 bytes of padding after `mValue` and another 2 bytes after `mNextAckTimeTick`.  If we reordered some members we could save 4 bytes.
* [ ] The pointer to the `ReliableMessageMgr` seems redundant: we can get at it via the `ExchangeContext` -> `ExchangeMgr` -> `ReliableMessageMgr` route.
* [ ] I wonder whether we could roll the reliable message delegate into the exchange delegate and save that pointer.
* [ ] Given our current allocation strategy (where a ReliableMessageContext is a by-value member of an ExchangeContext), the pointer to `EchangeContext` looks redundant to me.  We could get to it via `offsetof` and pointer arithmetic.
* [ ] I don't understand why some of the timeout tick values are uint32_t and some are uint16_t.  A 16-bit timeout in ms gets us to 65 seconds; I really doubt we expect CRMP timeouts larger than this....  And in practice we end up casting those 32-bit values to 16-bit various places, as far as I can see.

Fixing those issues, except for the "combine delegates" proposal, would save 16 bytes on 32-bit, as far as I can tell.

gerickson, 2021-04-16:
In terms of CRMP timeouts, should not those align with the potential sleep duration of a Thread SED node, which can be hours, days, or weeks?

yufengwangca, 2021-04-16:
The first retry timeout and active retry timeout can be adjusted by end device depending on its end. such as sleeping period. This is done by using Service Discovery Txt record CRA key

gerickson, 2021-04-16:
@yufengwangca, understood; however, I was responding to "A 16-bit timeout in ms gets us to 65 seconds; I really doubt we expect CRMP timeouts larger than this" above. SED sleep intervals may be far longer than 65 seconds, negotiated or not.

closed 2021-04-19 by andy31415

bzbarsky-apple, 2021-04-28:
@gerickson What is a practical situation in which the sending of, say, a CHIP command that does not complete within 10s of seconds would be followed by just resending it over and over instead of deciding there is an error?

The only sane way to handle devices that sleepy is to find out when they plan to wake up and transmit then....

gerickson, 2021-05-03:
> @gerickson What is a practical situation in which the sending of, say, a CHIP command that does not complete within 10s of seconds would be followed by just resending it over and over instead of [...]

I'll disagree that's the only sane way to do so. However, regardless, one use case is not that of a command request / response but rather a data model subscription state change notification flowing to the sleepy device from the publisher for which an ACK is returned in CRMP or no ACK in TCP.