project-chip/connectedhomeip #5896: Update ember library to have a better way to address a specific element of an attribute while reading/writing to it.

vivien-apple, 2021-04-09:
While working on #5020 and in order to minimise the amount of changes, an `index` argument has been added to some of the methods related to attributes reading/writing.

The goal of this argument is to address a particular element of the attribute. At the moment it is represented as an `int32` where `-1` represents read/write all the `List`, `0` represents read/write the actual number of elements in this list, and `n` read/write the nth element of the list.

Because of this actual element indexing starts at `1`. There should be a better way to express that.

Follow is Boris comment, asking for a followup.

Please document the semantics of this "index" arg and why we're adding "1" to it.  The fact that -1 becomes "read or write the length" and non-negative values become entry writes is _extremely_ non-obvious.

In terms of naming, readWritePartOfListAttribute (and similar for the separate writeAttribute/readAttribute methods) might be closer to what this thing really does.  And document really carefully, then file a followup to have a sane way of doing this.  :(  Will need some surgery on the ember stuff.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/5020#discussion_r610127406_

turon, 2022-02-01:
Referenced comment/code:

[src/app/clusters/descriptor/descriptor.cpp]
```
    record.manufacturerCode = EMBER_AF_NULL_MANUFACTURER_CODE;
    record.attributeId      = attributeId;

    return emAfReadOrWriteAttribute(&record, NULL, (uint8_t *) buffer, 0, write, index + 1);
```

``` 
@bzbarsky-apple bzbarsky-apple on Apr 8, 2021
Please document the semantics of this "index" arg and why we're adding "1" to it. The fact that -1 becomes "read or write the length" and non-negative values become entry writes is extremely non-obvious.

In terms of naming, readWritePartOfListAttribute (and similar for the separate writeAttribute/readAttribute methods) might be closer to what this thing really does. And document really carefully, then file a followup to have a sane way of doing this. :( Will need some surgery on the ember stuff.

Author @vivien-apple vivien-apple on Apr 9, 2021
#5896
```

turon, 2022-02-01:
The referenced code no longer calls `emAfReadOrWriteAttribute`.  
Marking stale to be closed after validation by author or reviewer.
Removing tag `v1_triage_split_4`

bzbarsky-apple, 2022-02-01:
The `index` arg went away in #12664.

closed 2022-02-01 by bzbarsky-apple