project-chip/connectedhomeip #5897: Passing write or readLength = 0 as an argument to emAfReadOrWriteAttribute is unsafe

vivien-apple, 2021-04-09:
Passing a 0 `readLength` here is just asking for trouble in memory safety terms for both reads and writes.  Please make sure a followup is filed to stop doing that, marked as a critical security problem.  Or fix it before landing, I guess....

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/5020#discussion_r610126472_

tcarmelveilleux, 2022-02-01:
A bit more context since the original PR conversation link does not expand:

[image]

bzbarsky-apple, 2022-02-01:
Note that the code in question (the caller of emAfReadOrWriteAttribute) is gone now.  The API itself still has issues, but this callsite is not one of them.

closed 2022-02-07 by woody-apple