project-chip/connectedhomeip #5926: Consider storing "fixed" attributes in .rodata only

bzbarsky-apple, 2021-04-10:
#### Problem
The CHIP spec defines a "F" quality for attributes which says they do not change short of firmware update.  This includes cluster revisions, for example, various info in Basic cluster, etc, etc.  All such attributes are readonly.

For such attributes, we should be able to skip generating space in the attribute store, and save some RAM.

 #### Proposed Solution
Add a way to flag attributes as "fixed", in Zap, only output default values for such attributes, stop allocating space in the attribute store for them.

@jepenven-silabs @jmartinez-silabs @vivien-apple

bzbarsky-apple, 2021-06-22:
One issue that was pointed out: fixed attributes in non-statically-allocated clusters (e.g. on a bridge) may not be easily storable in .rodata in our current setup.

woody-apple, 2022-02-01:
Moving to post V1.0, not clearly required for V1.0.